esa-tu-darmstadt/tapasco #23: Make synthesis and implementation effort configurable

sommerlukas, 2019-01-22:
The default settings used at the moment are AlternateRoutability + Retiming for Synthesis and Explore + PHYS_OPT_DESIGN for Implementation. These settings could be considered to be very high effort. A switch could be added to let the user decide between different "effort levels". For most synthesis runs it is not necessary to go with very high effort and the user might be happy about the much lower run-time.

Possible overlap with #21

sommerlukas, 2019-05-15:
@jahofmann, @cahz: From the TCL point of view, should the effort be a single variable that is evaluated in TCL to set different options for the tools or should we set a number of parameters for the tool (e.g. Vivado) directly from Scala?
In the latter case, what would these variables be and what would be reasonable values for them?
We probably would need to define a number of performance/optimization modes (cf. #21) and define the performance parameters for them.

jahofmann, 2019-05-15:
In the end we need to implement it in TCL.

I guess there are two concepts here. The frontend Tapasco should be kept clean from all Vivado specific optimizations. Instead we should provide some effort levels for synthesis and place/route like "low, medium, high, insane". Those can be translated to Vivado effort levels either in the Vivado Handler in Scala or TCL. Either way seems fine to me. As it is right now, doing it in TCL seems more Tapascoy.

sommerlukas, 2019-05-15:
Ok, I'll add the fronted interface and pass the desired effort level to TCL by filling a variable with a needle from Scala.

sommerlukas, 2019-05-21:
@cahz, @jahofmann: I've added an CLI option `--synthEffort` for the `compose`-command to allow users to define the desired synthesis and PnR effort level. 
There are six different modes available, see #21 for their names. 
They are passed on to `@@EFFORT_LEVEL@@` in `design.master.tcl.template`, as uppercase strings.

sommerlukas, 2019-06-11:
The functionality was added in #75;

closed 2019-06-11 by sommerlukas