esa-tu-darmstadt/tapasco #25: Evaluate 64 bit for platform_addr_t

sommerlukas, 2019-01-22:
All platforms except for legacy Zynq, such as the PCIe based systems or MPSoC, use larger than 32 bit addresses. While we currently get by with smaller addresses this might change in the future and we should consider a move to 64 bit addresses.

I currently don't see any problem just changing the address width. The Zynq platform should continue to work with the required casts and all other platforms currently cast to 64 bit addresses anyway.

sommerlukas, 2019-01-22:
Comment by @jkorinth:
Yes, sounds good. 👍 However, I'd like to test first if it would be feasible to simply use '`long` instead of `uint64_t`, which should have the native bitwidth of the platform. To be really transparent we'd also need to require little-endianess (so that the lower 32bit match on platforms, makes it simpler to develop hardware which works for both cases). Not sure if this would be problematic?

jahofmann, 2019-06-11:
Postponed until we do more in depth changes scheduled in 2019.10.

m-ober, 2019-08-10:
Working on 64-bit platform with FPGA memory > 4 GB:

```patch
diff --git a/runtime/arch/include/tapasco_types.h b/runtime/arch/include/tapasco_types.h
index 238b36c..ff662e9 100644
--- a/runtime/arch/include/tapasco_types.h
+++ b/runtime/arch/include/tapasco_types.h
@@ -62,7 +62,7 @@ typedef ul tapasco_job_id_t;
 #define PRIjob "%lu"
 
 /** Device memory location handle (opaque). **/
-typedef ul tapasco_handle_t;
+typedef uint64_t tapasco_handle_t;
 #define PRIhandle "%#08lx"
 
 /** default value for no flags **/
diff --git a/runtime/platform/include/platform_types.h b/runtime/platform/include/platform_types.h
index 03a2d3e..bb8f01e 100644
--- a/runtime/platform/include/platform_types.h
+++ b/runtime/platform/include/platform_types.h
@@ -58,7 +58,7 @@ typedef uint32_t platform_ctl_addr_t;
 #define PRIctl "%#08x"
 
 /** Device memory space address type (opaque). **/
-typedef uint32_t platform_mem_addr_t;
+typedef uint64_t platform_mem_addr_t;
 #define PRImem "%#08x"
 
 /** Identifies a slot in the design, i.e., a Function. **/
```

jahofmann, 2019-08-10:
The reason that fix is not integrated yet are legacy platforms like zynq. Those platforms support only 32 bit and might be a lot slower when forced to run on 64 bit.

All other platforms however are 64 bit, so it might be a tradeoff that's worthwhile.

A second solution would be integrating the size of memories into the status core and let the runtime choose an appropriate size. As the types are compile time, there has to be a way to change between the types at runtime. Otherwise there needs to be two versions of all those functions, one for 32 bit and one for 64 bit, which seems unclean...

jahofmann, 2019-11-05:
I've tested it on the ZC706 and there is no difference, check the attached graph:
[zc706.pdf](https://github.com/esa-tu-darmstadt/tapasco/files/3808988/zc706.pdf)

I will add the changes to my 32 bit fixes branch.

commit 60e1098c (2019-11-05): Use 64 bit memory address on all platforms

    - Fixes #25
    - Has no adverse impact on Zynq as shown in #25
closed 2019-11-05 by jahofmann