esa-tu-darmstadt/tapasco #26: Fallback option of requested amount of MSI-X interrupts is not available

sommerlukas, 2019-01-22:
The driver currently simply fails if the OS is not able/willing to provide the requested number of interrupts.

There should be a fall back option that gets enabled automatically if the requested amount of interrupts can not be provided.

sommerlukas, 2019-01-22:
Comment by @jahofmann:
My preferred solution would always use the last interrupt as a fall back interrupt. For example if 20 interrupts are available, interrupts 0 through 18 are dedicated and interrupt 19 is used to indicate that any of the other devices sent an interrupt. The driver provides coherent memory to the interrupt controller so the interrupt controller can send out a bit array of triggered interrupts before it sends the MSIx request. With the ordering rules enforced the host machine can simply check the words sent by the MSIx controller to see which device has fired.

cahz, 2024-04-15:
This is not an issue in reality, not on the roadmap anymore.

closed 2024-04-15 by cahz