esa-tu-darmstadt/tapasco #27: TaPaSCo is stuck after all jobs finished in verbose mode

sommerlukas, 2019-01-22:
When verbose-mode is activated (`-v`) and logs are tracked, the `MultiFileWatcher`s prevent TaPaSCo from exiting normally. Check that all watchers are properly terminated after their corresponding job has ended.

forflo, 2019-05-21:
I am having the same problem here, but also when omitting `-v`. I just did a `git pull` from master, rebuild everything with `sbt compile; sbt assembly`, and finally ran the line `tapasco hls arrayinit -p vcu118 --skipEvaluation`. The latency between `Tapasco finished successfully` and the actual termination is about 10 seconds which is relatively unbearble :( . Could you refer to places in the code so that I can investigate the issue myself? Thanks :)

forflo, 2019-05-21:
Sorry, I just measured the latency.
`time tapasco hls arrayinit -p vcu118 --skipEvaluation`
gives
`real	1m3.352s`.

lukasmweber, 2019-05-21:
This is a Problem with the Multi-Threading.

The problem occurs in the file `Tapasco.scala`, right in the last few lines (124-130).
In case any errors have occured, TaPaSCo is exited using `sys.exit(1)` in line 126.
In the else branch (when TapaSCo finished successfully), there is no `sys.exit(0)`.
I guess that this call kills the Runtime and thereby all Threads that are still running. The major problem here is determining if the remaining Threads do anything of importance or not.

This may also depend on what you are currently doing.
For testing purposes i added the `sys.exit(0)` call and it works, but i will have to further investigate if this call could potentially have unwanted side-effects.

sommerlukas, 2019-05-21:
@forflo: If you want to investigate that further, you can have a look at what the remaining threads are doing by looking at the output of lines 117-122 (e.g. by changing `logger.debug` to `logger.info`). 

@lukasmweber did some initial investigation, but any hints or help would be greatly appreciated.

jahofmann, 2019-05-27:
Is this solved with #64 ?

sommerlukas, 2019-05-27:
Yes, with the explicit shutdown of the execution context/threadpool, this should hopefully be fixed by now. The problem actually occurred independent of the verbose-mode.

closed 2019-05-27 by sommerlukas