esa-tu-darmstadt/tapasco #31: Allow PE Masters to have any valid AXI Data Width

sommerlukas, 2019-01-22:
The data width of PE masters is currently limited to either 32 or 64 bit. Considering that most platforms outside of Zynq have much broader memory controllers it is beneficial to support all valid AXI Data Widths up to 1024 bits. This might also be relevant for Zynq platforms if the designer of a PE wants to keep their logic simple and rely on data width converters to interface with the memories correctly.

sommerlukas, 2019-01-22:
Comment by @jahofmann:
Currently I'm testing a simple fix where the separation between 32 and 64 bit is removed in `arch_create_mem_interconnects`.

jkorinth, 2019-05-28:
That approach sounds good. I don’t think there’s any principal obstacle to use other widths; the split in the Tcl function was only a workaround for the time being, as I recall. There is some restriction on the AXI ICs, that makes grouping masters of same width necessary/preferable.

cahz, 2020-04-06:
This should not be an issue anymore. Please reopen if this still exists.

closed 2020-04-06 by cahz