esa-tu-darmstadt/tapasco #39: Add --skipEvaluation to HLS flow

jahofmann, 2019-01-22:
Add a parameter to skip evaluate similar to the feature implemented for the import flow.

forflo, 2019-03-12:
Does `--skipEvaluation` in this context mean that we only do RTL-Synthesis (without out-of-context synthesis). If yes, this would be a nice feature. Does anybody actually work on this at the moment?

sommerlukas, 2019-03-12:
Yes, the requested feature would work similar to the `--skipEvaluation`-flag for the `import`-command. That means it would run the C/C++-code through HLS and make the core available for `compose` without running the OOC-synthesis for evaluation.
I'm not yet actively working on that, but if you need the feature in time, I can try to give it priority and implement in the course of this week or so.

sommerlukas, 2019-03-13:
I've implemented the feature in 70c78eeb892b1f027a27abf71688bb0e5db7a897/ 5c5a45d6424b536bf4a625ef768fd769f258f935. 

With the `--skipEvaluation`-flag on an `hls` task, the automatic `import` following the actual HLS will skip the evaluation part.

closed 2019-03-13 by sommerlukas