esa-tu-darmstadt/tapasco #42: Fix VLNV for Vivado 2018.3

yannickl96, 2019-02-01:
When composing a design for vc709 I encounter the following error:
ERROR: [BD 5-216] VLNV <xilinx.com:ip:mig_7series:4.1> is not supported for the current part. The latest supported version for this part is:4.2
ERROR: [Common 17-39] 'create_bd_cell' failed due to earlier errors.

    while executing
"create_bd_cell -type ip -vlnv $vlnv $name"
    (procedure "tapasco::ip::create_mig_core" line 7)
    invoked from within
"tapasco::ip::create_mig_core $name"
    (procedure "create_mig_core" line 10)
    invoked from within
"create_mig_core "mig""
    (procedure "create_subsystem_memory" line 25)
    invoked from within
"create_subsystem_memory"
    ("eval" body line 1)
    invoked from within
"eval $cmd"
    ("foreach" body line 9)
    invoked from within
"foreach ss $sss {
      set name [string trim $ss "/"]
      set cmd  "create_subsystem_$name"
      puts "Creating subsystem $name ..."
      if {[ll..."
    (procedure "platform::create" line 12)
    invoked from within
"platform::create"

Just a reminder to fix the VLNV for Vivado 2018.3.

closed 2019-02-05 by jahofmann

jkorinth, 2019-05-28:
There will be a day when Xilinx relents and let’s us autodiscover the „latest“ VLNV of any core. *sigh*. Until then we‘ll have to fix the version specific Tcls with each iteration. Mmh, maybe someone could investigate whether there is a Tcl approach to automatically find the right version number yet? 🤔

cahz, 2019-05-28:
It is possible to leave out the version in a VLNV. Then we don't have to change it manually each time, however we won't notice changes in Xilinx IP cores.

commit a66b236d (2019-02-08): Adds VLNVs for Vivado 2018.3 MIG

    - Closes #42