esa-tu-darmstadt/tapasco #43: Build fails at target tapasco-benchmark

forflo, 2019-02-08:
## System Details
- 4.4.0-64-generic
- g++-5 (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
- g++-4.9 (Ubuntu 4.9.3-13ubuntu2) 4.9.3
- Ubuntu LTS 16.04
- Vivado installed (2018.2); in Path
- PATH="/home/demo/tapasco/bin:/home/demo/tapasco/build/install/usr/local/bin/:/home/demo/moreSpace/Vivado/2018.2/bin:/home/demo/tapasco/bin:/home/demo/tapasco/build/install/usr/local/bin/:/home/demo/moreSpace/Vivado/2018.2/bin:/home/demo/tapasco/bin:/home/demo/tapasco/build/install/usr/local/bin/:/home/demo/moreSpace/Vivado/2018.2/bin:/home/demo/.sdkman/candidates/sbt/current/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"

## Steps to reproduce
- cd ~/tapasco
- mkdir build && cd build
- cmake ..
- make

## Make output at fail

```
[ 90%] Generating json11/json11.cpp, json11/json11.hpp
Klone nach '/home/demo/tapasco/build/examples/tapasco-benchmark/json11' ...
remote: Enumerating objects: 299, done.
remote: Total 299 (delta 0), reused 0 (delta 0), pack-reused 299
Empfange Objekte: 100% (299/299), 82.88 KiB | 0 bytes/s, Fertig.
Löse Unterschiede auf: 100% (164/164), Fertig.
Prüfe Konnektivität ... Fertig.
Scanning dependencies of target tapasco-benchmark
[ 92%] Building CXX object examples/tapasco-benchmark/CMakeFiles/tapasco-benchmark.dir/tapasco_benchmark.cpp.o
In file included from /home/demo/tapasco/examples/tapasco-benchmark/tapasco_benchmark.cpp:34:0:
/home/demo/tapasco/arch/include/tapasco.hpp:30:2: error: #error "g++ 5.x.x or newer required (C++11 features)"
 #error "g++ 5.x.x or newer required (C++11 features)"
  ^
In file included from /home/demo/tapasco/examples/tapasco-benchmark/tapasco_benchmark.cpp:34:0:
/home/demo/tapasco/arch/include/tapasco.hpp: In constructor ‘tapasco::OutOnly<T>::OutOnly(T&)’:
/home/demo/tapasco/arch/include/tapasco.hpp:59:19: error: ‘is_trivially_copyable’ was not declared in this scope
     static_assert(is_trivially_copyable<T>::value, "Types must be trivially copyable!");
                   ^
/home/demo/tapasco/arch/include/tapasco.hpp:59:42: error: expected primary-expression before ‘>’ token
     static_assert(is_trivially_copyable<T>::value, "Types must be trivially copyable!");
                                          ^
/home/demo/tapasco/arch/include/tapasco.hpp:59:43: error: ‘::value’ has not been declared
     static_assert(is_trivially_copyable<T>::value, "Types must be trivially copyable!");
                                           ^
/home/demo/tapasco/arch/include/tapasco.hpp: In constructor ‘tapasco::InOnly<T>::InOnly(T&)’:
/home/demo/tapasco/arch/include/tapasco.hpp:79:19: error: ‘is_trivially_copyable’ was not declared in this scope
     static_assert(is_trivially_copyable<T>::value, "Types must be trivially copyable!");
                   ^
/home/demo/tapasco/arch/include/tapasco.hpp:79:42: error: expected primary-expression before ‘>’ token
     static_assert(is_trivially_copyable<T>::value, "Types must be trivially copyable!");
                                          ^
/home/demo/tapasco/arch/include/tapasco.hpp:79:43: error: ‘::value’ has not been declared
     static_assert(is_trivially_copyable<T>::value, "Types must be trivially copyable!");
                                           ^
/home/demo/tapasco/arch/include/tapasco.hpp: In constructor ‘tapasco::RetVal<T>::RetVal(T&)’:
/home/demo/tapasco/arch/include/tapasco.hpp:98:19: error: ‘is_trivially_copyable’ was not declared in this scope
     static_assert(is_trivially_copyable<T>::value, "Types must be trivially copyable!");
                   ^
/home/demo/tapasco/arch/include/tapasco.hpp:98:42: error: expected primary-expression before ‘>’ token
     static_assert(is_trivially_copyable<T>::value, "Types must be trivially copyable!");
                                          ^
/home/demo/tapasco/arch/include/tapasco.hpp:98:43: error: ‘::value’ has not been declared
     static_assert(is_trivially_copyable<T>::value, "Types must be trivially copyable!");
                                           ^
/home/demo/tapasco/arch/include/tapasco.hpp: In constructor ‘tapasco::Local<T>::Local(T&)’:
/home/demo/tapasco/arch/include/tapasco.hpp:110:19: error: ‘is_trivially_copyable’ was not declared in this scope
     static_assert(is_trivially_copyable<T>::value, "Types must be trivially copyable!");
                   ^
/home/demo/tapasco/arch/include/tapasco.hpp:110:42: error: expected primary-expression before ‘>’ token
     static_assert(is_trivially_copyable<T>::value, "Types must be trivially copyable!");
                                          ^
/home/demo/tapasco/arch/include/tapasco.hpp:110:43: error: ‘::value’ has not been declared
     static_assert(is_trivially_copyable<T>::value, "Types must be trivially copyable!");
[... 218 lines cut ...]
In file included from /usr/include/c++/4.9/bits/stl_algobase.h:67:0,
                 from /usr/include/c++/4.9/bits/char_traits.h:39,
                 from /usr/include/c++/4.9/ios:40,
                 from /usr/include/c++/4.9/ostream:38,
                 from /usr/include/c++/4.9/iostream:39,
                 from /home/demo/tapasco/examples/tapasco-benchmark/tapasco_benchmark.cpp:26:
/usr/include/c++/4.9/bits/stl_iterator.h:328:5: note: template<class _Iterator> typename std::reverse_iterator<_Iterator>::difference_type std::operator-(const std::reverse_iterator<_Iterator>&, const std::reverse_iterator<_Iterator>&)
     operator-(const reverse_iterator<_Iterator>& __x,
     ^
/usr/include/c++/4.9/bits/stl_iterator.h:328:5: note:   template argument deduction/substitution failed:
In file included from /home/demo/tapasco/examples/tapasco-benchmark/tapasco_benchmark.cpp:42:0:
/home/demo/tapasco/examples/tapasco-benchmark/TransferSpeed.hpp:62:42: note:   ‘std::chrono::_V2::system_clock::time_point {aka std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long int, std::ratio<1l, 1000000000l> > >}’ is not derived from ‘const std::reverse_iterator<_Iterator>’
       d = high_resolution_clock::now() - tstart;
                                          ^
/home/demo/tapasco/examples/tapasco-benchmark/tapasco_benchmark.cpp: In function ‘int main(int, const char**)’:
/home/demo/tapasco/examples/tapasco-benchmark/tapasco_benchmark.cpp:182:45: error: ‘put_time’ was not declared in this scope
     str << put_time(&tm, "%Y-%m-%d %H:%M:%S");
                                             ^
examples/tapasco-benchmark/CMakeFiles/tapasco-benchmark.dir/build.make:70: die Regel für Ziel „examples/tapasco-benchmark/CMakeFiles/tapasco-benchmark.dir/tapasco_benchmark.cpp.o“ scheiterte
make[2]: *** [examples/tapasco-benchmark/CMakeFiles/tapasco-benchmark.dir/tapasco_benchmark.cpp.o] Fehler 1
CMakeFiles/Makefile2:573: die Regel für Ziel „examples/tapasco-benchmark/CMakeFiles/tapasco-benchmark.dir/all“ scheiterte
make[1]: *** [examples/tapasco-benchmark/CMakeFiles/tapasco-benchmark.dir/all] Fehler 2
Makefile:149: die Regel für Ziel „all“ scheiterte

```

jahofmann, 2019-02-08:
It seems like you're using GCC 4.9 which does not support C++11 features.

GCC 5 or newer is required to build the run time.

forflo, 2019-02-08:
Ok, nice.

`CC=gcc-5 CXX=g++-5 cmake ..`

fixed it.

closed 2019-02-08 by forflo