esa-tu-darmstadt/tapasco #45: $TAPASCO_HOME/common/common.tcl triggered exit in vivado_hls

forflo, 2019-02-08:
## System Details
- 4.4.0-64-generic
- g++-5 (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
- g++-4.9 (Ubuntu 4.9.3-13ubuntu2) 4.9.3
- Ubuntu LTS 16.04
- Vivado installed (2018.2); in Path
- PATH="/home/demo/tapasco/bin:/home/demo/tapasco/build/install/usr/local/bin/:/home/demo/moreSpace/Vivado/2018.2/bin:/home/demo/tapasco/bin:/home/demo/tapasco/build/install/usr/local/bin/:/home/demo/moreSpace/Vivado/2018.2/bin:/home/demo/tapasco/bin:/home/demo/tapasco/build/install/usr/local/bin/:/home/demo/moreSpace/Vivado/2018.2/bin:/home/demo/.sdkman/candidates/sbt/current/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"

## Problem Summary

`tapasco -v compose [ arrayinit x 4 ] @ 100 MHz -p vcu118` fails because Vivado HLS finishes with non-zero exit code (1):

```
[15:46:48 <main: Tapasco$> INFO] Running with configuration: [Configuration @/home/demo/tapasco/default.cfg]
Verbose = Some(verbose)
KernelDir = /home/demo/tapasco/kernel
CoreDir = /home/demo/tapasco/core
ArchDir = /home/demo/tapasco/arch
PlatformDir = /home/demo/tapasco/platform
Slurm = false
Parallel = false
MaxThreads = unlimited
MaxTasks = unlimited
Jobs = List(ComposeJob([arrayinit x 4],100.0,Vivado,None,Some(ArrayBuffer(vcu118)),None,None))
[15:46:48 <main: Compose$> INFO] need to synthesize the following cores first: arrayinit @ axi4mm@vcu118
[15:46:48 <pool-1-thread-1: VivadoHighLevelSynthesis$> INFO] starting run 'arrayinit' for axi4mm@vcu118: output in /home/demo/tapasco/core/arrayinit/axi4mm/vcu118/hls/axi4mm.log
[15:46:48 <pool-1-thread-1: VivadoHighLevelSynthesis$> INFO] verbose mode verbose is active, starting to watch /home/demo/tapasco/core/arrayinit/axi4mm/vcu118/hls/axi4mm.log
[15:46:50 <pool-1-thread-1: VivadoHighLevelSynthesis$> INFO] Script name was: /home/demo/tapasco/core/arrayinit/axi4mm/vcu118/hls/axi4mm.tcl
[15:46:50 <pool-1-thread-1: VivadoHighLevelSynthesis$> INFO] Logfile name was: /home/demo/tapasco/core/arrayinit/axi4mm/vcu118/hls/axi4mm.log
[15:46:50 <pool-1-thread-1: VivadoHighLevelSynthesis$> INFO] Pwd was: /home/demo/tapasco/core/arrayinit/axi4mm/vcu118/hls
[15:46:50 <pool-1-thread-1: VivadoHighLevelSynthesis$> ERROR] Vivado HLS finished with non-zero exit code: 1 for 'arrayinit' for axi4mm@vcu118
[15:46:50 <main: HighLevelSynthesis$> INFO] all HLS tasks have finished.
[15:46:50 <main: Compose$> ERROR] HLS tasks failed, aborting composition
[15:46:50 <main: Tapasco$> ERROR] TaPaSCo finished with errors
```

Note, that vivado complains about not knowing how to deal with a `-notrace`. 

## Problem Isolation (after first run of tapasco compose [...])

- `tapasco -v compose [ arrayinit x 4 ] @ 100 MHz -p vcu118` -> fails
- `cd /home/demo/tapasco/core/arrayinit/axi4mm/vcu118/hls` -> change into path where vivado_hls had been executed before
- `vivado_hls -f /home/demo/tapasco/core/arrayinit/axi4mm/vcu118/hls/axi4mm.tcl` -> fails
- replace lines 25 - 27 in `home/demo/tapasco/common` with: 
    ```tcl
  proc source_quiet {fn} {
    eval "source [expr {[string is space [info commands version]] ? {} : {}}]" $fn
  }
    ```
- `vivado_hls -f /home/demo/tapasco/core/arrayinit/axi4mm/vcu118/hls/axi4mm.tcl` -> succeeds
- `tapasco -v compose [ arrayinit x 4 ] @ 100 MHz -p vcu118` -> succeeds

forflo, 2019-02-08:
I didn't do a PR, because I don't know TCL well enough to provide for an appropriate patch.

jahofmann, 2019-02-08:
Could you attach the content of the file /home/demo/tapasco/core/arrayinit/axi4mm/vcu118/hls/axi4mm.log?

forflo, 2019-02-08:
The content of the file you requrested:

```

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.2 (64-bit)
  **** SW Build 2258646 on Thu Jun 14 20:02:38 MDT 2018
  **** IP Build 2256618 on Thu Jun 14 22:10:49 MDT 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source /home/demo/moreSpace/Vivado/2018.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/home/demo/moreSpace/Vivado/2018.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'demo' on host 'ubuntu' (Linux_x86_64 version 4.4.0-64-generic) on Fri Feb 08 15:36:45 CET 2019
INFO: [HLS 200-10] On os Ubuntu 16.04.5 LTS
INFO: [HLS 200-10] In directory '/home/demo/tapasco/core/arrayinit/axi4mm/vcu118/hls'
ERROR: [HLS 200-101] 'source': Unknown option '-notrace'.

FORMAT
  source [<TCL file name>]
      <string>
    -encoding <string>

command 'ap_source' returned error code
    while executing
"source -notrace /home/demo/tapasco/common/subsystem.tcl"
    ("eval" body line 1)
    invoked from within
"eval "source [expr {[string is space [info commands version]] ? {} : {-notrace}}]" $fn"
    (procedure "source_quiet" line 2)
    invoked from within
"source_quiet $::env(TAPASCO_HOME)/common/subsystem.tcl"
    (in namespace eval "::tapasco" script line 7)
    invoked from within
"namespace eval tapasco {
  namespace export source_quiet
  proc source_quiet {fn} {
    eval "source [expr {[string is space [info commands version]] ..."
    (file "/home/demo/tapasco/common/common.tcl" line 23)
    invoked from within
"source /home/demo/tapasco/common/common.tcl"
    (file "/home/demo/tapasco/core/arrayinit/axi4mm/vcu118/hls/axi4mm.tcl" line 25)
    invoked from within
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Fri Feb  8 15:36:45 2019...
```

jahofmann, 2019-02-08:
That bug should've been fixed in commit 1c56c3458ab3ffd7770975f4f64d90f22ae975b2
Which version are you running?

closed 2019-02-08 by forflo