esa-tu-darmstadt/tapasco #46: Add board part repository to platform configuration

m-ober, 2019-02-09:
It would be nice to have an additional parameter in the `platform.json` to configure a board part repository. The resulting code should then look like this (new code between `create_project` and `if`).

```tcl
# setup the project
create_project microarch [pwd]/microarch -part {xcvu9p-flgb2104-2-i} -force

set urepo1 $::env(AWS_FPGA_REPO_DIR)/hdk/common/shell_v04261818/hlx/design/boards
set_property BOARD_PART_REPO_PATHS ${urepo1} [current_project]

if {"xilinx.com:f1_cl:part0:1.0" != ""} {
  set_property board_part {xilinx.com:f1_cl:part0:1.0} [current_project]
}
```

sommerlukas, 2019-02-11:
How would the corresponding `platform.json` look like in your example? 
Which part of the `urepo1`-definition results from the `platform.json`? If the `env(...)` should also come from JSON, we would need a syntax to mark parts of the repository URL as environment variables.

jahofmann, 2019-02-11:
@sommerlukas Thanks for Responding, but the Issue was requested by me as a reminder and I've got the fix coming :)

m-ober, 2019-02-18:
Thanks for implementing, unfortunately it does not seem to work (for me).

This is my `platform.json`:

```json
{
	"Name" : "aws",
	"Description" : "AWS F1 Platform",
	"TclLibrary" : "aws.tcl",
	"Part" : "xcvu9p-flgb2104-2-i",
	"BoardPartRepository" : "$::env(AWS_FPGA_REPO_DIR)/hdk/common/shell_v04261818/hlx/design/boards",
	"BoardPart" : "xilinx.com:f1_cl:part0:1.0",
	"BoardPreset" : "f1_cl",
	"TargetUtilization" : 90,
	"Benchmark" : "",
	"HostFrequency": 250.0,
	"MemFrequency": 200.0
}
```

Which produces this code (repository is an empty string):

```tcl
create_project microarch [pwd]/microarch -part {xcvu9p-flgb2104-2-i} -force
if {"xilinx.com:f1_cl:part0:1.0" != ""} {
  if {"" != ""} {
    set_property BOARD_PART_REPO_PATHS {} [current_project]
  }
  set_property board_part {xilinx.com:f1_cl:part0:1.0} [current_project]
}
```

m-ober, 2019-02-18:
Furthermore, I think we need `"@@BOARD_PART_REPOSITORY@@"` instead of `{@@BOARD_PART_REPOSITORY@@}`, because the curly braces disable substitution [1] and thus the `$::env` command cannot be used.

[1] https://www.tcl.tk/man/tcl8.5/tutorial/Tcl4.html

cahz, 2019-02-18:
The replacement is working for me after adding escape characters (`\\$`) Have you recompiled TaPaSCo after the update by running `sbt compile assembly`?

m-ober, 2019-02-18:
Thanks for the hint :wink: After recompiling tapasco, it's "almost" working – As written in https://github.com/esa-tu-darmstadt/tapasco/issues/46#issuecomment-464705772, I have to replace the `{}` with `""` in order to get it working, because otherwise the `$::env` command is not evaluated and the repository is not set correctly:

> ERROR: [Board 49-71] The board_part definition was not found for xilinx.com:f1_cl:part0:1.0

cahz, 2019-02-18:
Pushed the change in d5de076.

commit 5fbdcddc (2019-02-11): Add parameter to set a custom board repository path

    - Paramter is boardPartRepository in JSON
    - Closes #46
closed 2019-02-18 by cahz