esa-tu-darmstadt/tapasco #47: Properly report incorrect kernel.json files

forflo, 2019-02-11:
`tapasco -v hls mytest -p vcu118` does not produce a helpful error message on the following json:
```json
{
    "Description" : "Matrix multiplication kernel",
    "Name" : "mytesu",
     ...
    "Files" : ["mytest.c"],
    "Arguments" : [
       ...
    ]
}
```

```
[15:49:00 <main: Tapasco$> ERROR] no valid Kernels selected! (available: warraw, mytesu, myarray, arrayinit, arrayupdate, arraysum, counter, sudoku, rot13)
```

- Maybe add a check for correct semantics?

sommerlukas, 2019-02-11:
I'm not sure, whether the JSON is actually the problem here. You are requesting the kernel "mytest" in the command-line, but the kernel is name "mytesu" (Note the u instead of t here). 
The error message given by TaPaSCo seems to make sense to me, in particular as it even lists the kernel with the name given in the JSON (third item in the list of available kernels).

sommerlukas, 2019-03-14:
@forflo: Did you have a chance to test that in the meantime?

forflo, 2019-03-14:
Yes. I knew, how to fix it. However, I was slightly disappointed with tapasco's error message not pointing out what you wrote in your comment (from 11 Feb). I would wish for an error message like `the json key "name" does not match the directory name`. Sorry, if I was being unclear.

Another Question tough:
Is it somewhere documented that the `"Id"` key of a kernel.json cannot contain the number 0 as id?

Test code: 
```json
{
  "Arguments": [
    {
      "Name": "a",
      "Passing": "by reference"
    },
    {
      "Name": "b",
      "Passing": "by reference"
    },
    {
      "Name": "resp__",
      "Passing": "by reference"
    }
  ],
  "Description": "Orka kernel code for kernel 0",
  "Files": [
    "ip_0.c"
  ],
  "Id": 0,
  "Name": "Ip0",
  "TopFunction": "ip",
  "Version": "0.1"
}
```

which does produce the error
`[16:59:21 <main: FileAssetManager$$anon$2> WARN] could not build Kernel from file: '/home/demo/orka/ExplorerIps/ExplorerIp0/kernel.json'`

when doing
`tapasco -v --coreDir $(pwd)/Cores --kernelDir $(pwd)/Ips/ hls Ip0 -p vcu118`

jahofmann, 2019-03-14:
Currently Tapasco assumes that a slot is empty if the ID is 0, thus, ID 0 is not supported right now. Any particular reason you'd need the ID to be 0?

sommerlukas, 2019-03-14:
> Yes. I knew, how to fix it. However, I was slightly disappointed with tapasco's error message not pointing out what you wrote in your comment (from 11 Feb). I would wish for an error message like the json key "name" does not match the directory name. Sorry, if I was being unclear.

There might be cases, where the user intents to not match the name of the directory, in which the kernel-description resides, and the kernel name. For example, if one would generate multiple kernels from a single C source-file, the kernel-names won't match the directory name intentionally. Therefore, IMHO it's not a good idea to throw an error/print a warning if the kernel name and the directory name do not match.

sommerlukas, 2019-06-25:
For release 2019.10, we will rework the parser for `kernel.json`-files. The new parser should give better and more informative error messages to allow users to easily identify errors in their files.

PR #151 (2019-11-12): Feature/json parser error messages
This MR fixes the following Problems:

* JSON-Error Reporting: Json-Files are now handled with more care. Especially Errors in `kernel.json` Files are now reported with specific error messages.
* Duplicated Error Messages: Missing AverageClockCycle-Values often lead to 10s to 100s of Errors,  since it was reported each time the file was handled. Now it is only reported once per File.

Closes #47.

Wait for Pipeline to finish before Merging: [Pipeline 1626](https://git.esa.informatik.tu-darmstadt.de/tapasco/tapasco/pipelines/1626)
closed 2019-11-12 by cahz