esa-tu-darmstadt/tapasco #48: Don't choke on patched Vivado versions

m-ober, 2019-03-01:
Currently, when using a patched Vivado version, Tapasco fails at least at two places:

> Could not find /scratch/mio/tapasco_f1/common/common_2018.2_AR71715.tcl, Vivado 2018.2_AR71715 is not supported yet! 

While this can easily be resolved by placing a symlink to the `common_2018.2.tcl` file, there is also a problem with the `tapasco_status.json` which causes Tapasco (or rather the JSON Parser) to throw an Exception and exit:

```
   "Versions"              : [{
        "Software" : "Vivado",
        "Year"     : 2018_AR71715,
        "Release"  : 2_AR71715
```

Results in:
> com.fasterxml.jackson.core.JsonParseException: Unexpected character ('_' (code 95))

sommerlukas, 2019-03-01:
Can you test if the following patch fixes the second problem for you:
```
diff --git a/common/ip.tcl b/common/ip.tcl
index d4ea349..1ce5f05 100644
--- a/common/ip.tcl
+++ b/common/ip.tcl
@@ -478,7 +478,7 @@ namespace eval ::tapasco::ip {
     }
     puts "  finished address map, composing JSON ..."

-    set regex {([0-9][0-9][0-9][0-9]).([0-9][0-9]*)}
+    set regex {([0-9][0-9][0-9][0-9]).([0-9][0-9]*)(?:_.*)?}
     set no_intc [::platform::number_of_interrupt_controllers]
     set ts [clock seconds]
```
The extension to the regex will ignore the patch-specific part of the version number (`_AR71715` in your case). We cannot store this part in the status core anyway, because the registers hold only numbers and not strings.

m-ober, 2019-03-01:
Yes, this fixes the second problem 👍

m-ober, 2019-03-01:
It should probably be noted that there are even more possibilities for "non standard" version strings, e.g.

> Vivado v2018.2_AR71275_op (64-bit)
> Vivado v2018.2.op (64-bit)

I.e. there is a dot instead of an underline after the actual version number, or there are multiple patches applied.

sommerlukas, 2019-03-03:
For such cases, the patch could be extended to not break with the version number extensions:
```
diff --git a/common/ip.tcl b/common/ip.tcl
index d4ea349..1ce5f05 100644
--- a/common/ip.tcl
+++ b/common/ip.tcl
@@ -478,7 +478,7 @@ namespace eval ::tapasco::ip {
     }
     puts "  finished address map, composing JSON ..."

-    set regex {([0-9][0-9][0-9][0-9]).([0-9][0-9]*)}
+    set regex {([0-9][0-9][0-9][0-9]).([0-9][0-9]*)(?:_.*)?}
     set no_intc [::platform::number_of_interrupt_controllers]
     set ts [clock seconds]
```
However, as this patch basically only ignores the additional information, so TaPaSCo won't break with patched Vivado versions, and does not store the data. Therefore, @cahz and I decided we won't add that fix to the mainline TaPaSC. 

I will close the issue for now.

m-ober, 2019-03-17:
Sorry, but I have to disagree with closing this issue. Currently, it is not possible to use Tapasco out of the box with the Amazon FPGA Developer AMI, because it uses a patched Vivado version by default:

> [centos@ip-172-31-47-146 ~]$ vivado -version
> Vivado v2018.2_AR71275_op (64-bit)

While I think this is a low-priority issue for now, it's still an issue (unless Tapasco is not supposed to be run on AWS) :wink:

sommerlukas, 2019-03-17:
IMHO, adding this patch to "mainline" TaPaSCo would suggest that we fully support this scenario, which we don't because we cannot store the additional information in the status core currently.

From my point of view, it's ok that working with a *patched* version of Vivado also implies to work with a *patched* version of TaPaSCo.

@jahofmann: What do you think?

sommerlukas, 2019-05-15:
@jahofmann: What do you think, should we have this fix in 'mainline' TaPaSCo, even though we cannot store the additional information in the status core or should this remain a patch, that affected users have to apply?

jahofmann, 2019-05-15:
We should simply add the patch version information to the status core. Store the major version e.g. 2018.2 as it is and add a reference to a string at some other location in the status core. That can be a simple tuple of (2 byte length, extra version information string).

sommerlukas, 2019-05-15:
Ok, then let's first implement support for that in the status core (you wanted to rework that one anyway, didn't you?).

commit c256cc3a (2019-07-02): Adds support for patched versions of Vivado

    - Closes #48
PR #105 (2019-07-05): Replace hard coded status core with dynamic version
 * Replaces the custom, hard-coded status core format with a protobuf version
 * Easily add information to the status core and use that information from the runtime and driver
 * Adds necessary dependencies
 * Custom statically-linked rust executable is used to translate the JSON generated by the TaPaSCo flow to a protobuf binary
 * All of the examples are fixed and tested
 * The memory map as seen from user space is now completely platform independent. The driver is used to translate the virtual addresses to device specific ones.
 * Closes # [... cut]
closed 2019-10-07 by sommerlukas