esa-tu-darmstadt/tapasco #53: DSE: Delete projects for failed runs.

yannickl96, 2019-04-12:
TaPaSCo should delete the projects of failed compositions in the DSE so the file system is not filling up with junk.

jkorinth, 2019-05-18:
I think there has been a command switch for that already, part of the DSE command. It keeps the logs around, but deletes the huge Vivado projects and PnR data.

m-ober, 2019-05-28:
What if I want to analyze a failed run, e.g. look at the riming report in Vivado?

Maybe it would be good to have a clean-up procedure (platform specific), that removes the bitstream, DCPs and maybe other files.

jahofmann, 2019-05-28:
You can analyze these issues through simple compose runs. When you go through DSE it is assumed that you have at least done compose runs before and checked that your design performs to your liking.

lukasmweber, 2019-05-28:
Im currently implementing this functionalit and my take on this is the introduction of a flag `--deleteProjects` for `compose` and `explore` jobs, so that the user can specify whether they want to keep these files (like it is now) or if all project files should be discarded (if the flag is set).

sommerlukas, 2019-06-07:
The requested functionality was added in #72.

commit a1f392c6 (2019-06-07): Added Testing and fixed typos; Closes #53
closed 2019-06-07 by sommerlukas