esa-tu-darmstadt/tapasco #55: Parser does not handle large files properly

jahofmann, 2019-05-09:
[#8361](https://git.esa.informatik.tu-darmstadt.de/tapasco/tapasco/-/jobs/8361) failed for 2cd0d3a0541ee395aaa4f24e8bf442c8aaa64828:

`[16:16:59 <pool-5-thread-2-ScalaTest-running-JobParsersSpec: Common$> ERROR] parser exception: expected End at 32676`

Most likely the buffers are too small and larger files are not handled properly...

sommerlukas, 2019-05-13:
The problem actually is not the size of the files, but rather the fact that the testcases use hls commands with kernel names that start with `all` such as `all4rv...`, which is not allowed in TaPaSCo.

sommerlukas, 2019-05-17:
a4e0848045ffc00fb47f95e04f96e85e584496ce fixes the problem with the test: The randomly generated kernel names now do not contain "all" as substring anymore.

closed 2019-05-17 by sommerlukas