esa-tu-darmstadt/tapasco #56: Seperate Runtime and Bitstream Generation in Directory Structure

jahofmann, 2019-05-15:
This change would make it easier to see what parts for Tapasco are relevant for Bitstream generation and what parts are relevant for executing a bitstream on a target device.

Can be done as:
 * Separate Repositories
 * Directory Structure such as:
     - Tapasco
        - Bitstream
        - Runtime

Related issues #40

jahofmann, 2019-05-15:
@cahz @sommerlukas 
Any preference?

sommerlukas, 2019-05-15:
I agree, it would be nice to separate the tool-flow and the runtime more. 
Regarding the actual implementation, I would prefer to just separate the directories. Having multiple repositories IMHO has a number of backdraws:
* Much more complicated installation process
* Users always need to keep two repositories up-to-date, otherwise the different versions of the repositories might not work correctly with each other.
* Even after separating the two parts, their might be some interdependencies between them. So they might still need to know about each other, e.g. through environment variables, which makes running TaPaSCo unneccessarily complicated. 

Besides that, I do not see any real benefit in separating TaPaSCo into two separate repositories.

sommerlukas, 2019-06-25:
We have split the folder structure into `runtime` and `toolflow` in 0ca6174746c11afef9193ae32b600ff74a153055. 

In the course of the restructuring, `sbt` was replaced with `gradle` and the GUI `itapasco` was completely removed.

closed 2019-06-25 by sommerlukas