esa-tu-darmstadt/tapasco #59: Write proper status information during compose runs

jahofmann, 2019-05-16:
Currently the user is completely unaware of the progress of a compose run. Knowing the log files or using external tools such as vivado_progress can somewhat help experienced users but I'd suggest that Tapasco itself prints out useful information.

I propose the following states:
 - Composition generation
 - Synthesis
 - Place and Route
 - Write bitstream

Each stage should have timers that show the elapsed time for the current stage and the total runtime.

The stages can be parsed form the log files by using e.g. the plugin infrastructure to mark the stages with unique strings. This approach avoids changed log files in newer Vivado versions breaking this feature.

jkorinth, 2019-05-18:
There has been a log follow switch for that - you can make TaPaSCo follow multiple logs automatically, and open subsequent logs with the help of some regexes.

jahofmann, 2019-05-18:
The point of this issue is to not follow the huge amount of detail in the logs but still get some kind status information about the status of the run, especially for inexperienced users.

lukasmweber, 2019-05-27:
This Issue  Was fixed by Pull Request #66 .

closed 2019-05-27 by lukasmweber