esa-tu-darmstadt/tapasco #67: Wrong time format for status information

m-ober, 2019-05-27:
I suppose the format of the time is hh:mm:ss. But then "0:18:1108" does not make sense.

> [18:22:48 <Thread-6: VivadoComposer> INFO] Started System Composition (Total Elapsed: 0:00:04)
> [18:41:16 <Thread-6: VivadoComposer> INFO] Finished System Composition after 0:18:1108 (Total Elapsed: 0:18:1112)
> [18:41:16 <Thread-6: VivadoComposer> INFO] Started Synthesis (Total Elapsed: 0:18:1112)

jkorinth, 2019-05-28:
It is not a timestamp, the timestamp occurs right at the beginning, in the format you mentioned. 0:18:1112 is a duration of 18.1112 minutes in the synthesis step.

jahofmann, 2019-05-28:
@jkorinth That's a new feature introduced by lukasweber and already fixed in 08de6f0c471b9e9e55701b4f1721d480203e9db5

closed 2019-05-28 by jahofmann