esa-tu-darmstadt/tapasco #69: Revert back to Interconnects until problems with SmartConnects are fixed

jahofmann, 2019-05-28:
Problems like https://www.xilinx.com/support/answers/70838.html might make SmartConnect still inferior to Interconnects. To avoid such problems (as well as performance problems we experienced e.g. on NDP or Network applications). We should go back to Interconnects.

Instead of simply reverting the appropriate changes to move to interconnects we should instead create a nice capsulated IC generator that improves code quality in the corresponding areas.

jkorinth, 2019-05-28:
😤Are you saying my awesome Tcl generator needs improvement?! 😂

jkorinth, 2019-05-28:
But, seriously, you’re right, the idea was always to make much more use of external „plugin-like“ code generators, like the status core, just neater. And get rid of as much Tcl as possible/sensible. How do you implement such generators nowadays, still Chisel? Or is Bluespec also supported? I think it would be good to make the skeletal approach more visible, i.e., to have literal representations of the abstract  building blocks (like the „aggregators“) in the code, where a plugin can be called to generate the module.

jahofmann, 2019-06-11:
Move to Algun dia as this issue requires a better memory testing infrastructure.