esa-tu-darmstadt/tapasco #80: Add 32bit Zynq build for tlkm to CI

cahz, 2019-06-11:


cahz, 2019-11-11:
Linux headers from a x86_64 system are not working for this task and the compiler does not find all header files. To be able to cross-compile the kernel module, I have not found a solution without cross-compiling the Linux kernel before. Due to the complexity I wouldn't do this in a CI runner.

jahofmann, 2019-11-11:
The runtime libraries should be fine, though.

cahz, 2019-11-11:
Runtime libraries are now in CI (on branch feature/crosscompiling). It would be nice to also detect problems in the kernel module (e.g. fixes in 576ba4068f62095c804e879c4a928fb04431486d).

jahofmann, 2019-11-11:
I think we can add an extra runner just for running 32 bit applications. This runner would run on a dedicated 32 bit VM. That shouldn't be too difficult.

PR #153 (2019-11-12): Add support for cross-compilation and clang compiler
Add options to change compiler and cross-compile the TaPaSCo platform library. Build tlkm kernel module on 32bit docker image to close #80.
closed 2019-11-12 by cahz