esa-tu-darmstadt/tapasco #85: --maxTasks setting is one off (or has misleading documentation)

m-ober, 2019-06-21:
Starting TaPaSCo with

`tapasco --maxThreads 1 --maxTasks 4 explore [vexriscv_pe x 1] in frequency`

launches 3 (not 4) compose tasks, i.e.:

> [15:32:43 <pool-1-thread-1: ConcreteExploration> INFO] current batch: 0 [12]
> [15:32:43 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=450.000] for axi4mm@aws
> [...]
> [15:32:44 <pool-1-thread-2: VivadoComposer> INFO] Vivado starting run 0x965e5fcc[vexriscv_pe x 1] with axi4mm@aws[F=450.000]: show progress with `vivado_progress /scratch/mio/tapasco_f1/DSE_2019-06-21T15-32-43.877902/bd/axi4mm/aws/vexriscv_pe/001/450.0/axi4mm-aws--vexriscv_pe_1--450.0.log`
> [15:32:44 <pool-1-thread-3: VivadoComposer> INFO] Vivado starting run 0x965e5fcc[vexriscv_pe x 1] with axi4mm@aws[F=445.000]: show progress with `vivado_progress /scratch/mio/tapasco_f1/DSE_2019-06-21T15-32-43.877902/bd/axi4mm/aws/vexriscv_pe/001/445.0/axi4mm-aws--vexriscv_pe_1--445.0.log`
> [15:32:44 <pool-1-thread-4: VivadoComposer> INFO] Vivado starting run 0x965e5fcc[vexriscv_pe x 1] with axi4mm@aws[F=440.000]: show progress with `vivado_progress /scratch/mio/tapasco_f1/DSE_2019-06-21T15-32-43.877902/bd/axi4mm/aws/vexriscv_pe/001/440.0/axi4mm-aws--vexriscv_pe_1--440.0.log`
> [15:32:46 <Thread-6: VivadoComposer> INFO] Started System Composition (Total Elapsed: 0:00:02)
> [15:32:46 <Thread-11: VivadoComposer> INFO] Started System Composition (Total Elapsed: 0:00:02)
> [15:32:46 <Thread-16: VivadoComposer> INFO] Started System Composition (Total Elapsed: 0:00:02)
> 

The help text says:

>   --maxTasks NUM                Limit max. number tasks executed in parallel to 
>                                 the given number

So I think one more thread should be launched or the help text changed to something like "Limit max. number tasks executed in parallel to the given number **(note: this includes the main TaPaSCo thread, so NUM - 1 compute tasks are started)**"

sommerlukas, 2019-06-24:
It's more a problem of the documentation, IMHO the behavior is correct. We will extend the documentation.

lukasmweber, 2019-06-26:
This issue was closed by #94.

closed 2019-06-26 by lukasmweber