esa-tu-darmstadt/tapasco #92: DSE does not stop after working frequency found

m-ober, 2019-06-25:
Starting a DSE in one dimension (frequency), the following batch job was started:

> [22:44:00 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=355.000] for axi4mm@aws                                                                                       
> [22:44:00 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=350.000] for axi4mm@aws                                                                                       
> [22:44:00 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=345.000] for axi4mm@aws                                                                                       
> [22:44:00 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=340.000] for axi4mm@aws                                                                                       
> [22:44:00 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=335.000] for axi4mm@aws                                                                                       
> [22:44:00 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=330.000] for axi4mm@aws                                                                                       
> [22:44:00 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=325.000] for axi4mm@aws                                                                                       
> [22:44:00 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=320.000] for axi4mm@aws                                                                                       
> [22:44:00 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=315.000] for axi4mm@aws                                                                                       
> [22:44:00 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=310.000] for axi4mm@aws                                                                                       
> [22:44:00 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=305.000] for axi4mm@aws                                                                                       
> [22:44:00 <pool-1-thread-1: ConcreteBatch> INFO] starting [[vexriscv_pe x 1]] [F=300.000] for axi4mm@aws                                                                                       
> [22:44:00 <pool-1-thread-2: VivadoComposer> INFO] Vivado starting run 0x965e5fcc[vexriscv_pe x 1] with axi4mm@aws[F=355.000]: show progress with `vivado_progress /scratch/mio/tapasco_f1/DSE_2
> 019-06-21T22-44-00.422707/bd/axi4mm/aws/vexriscv_pe/001/355.0/axi4mm-aws--vexriscv_pe_1--355.0.log`                                                                                            
> [22:44:00 <pool-1-thread-3: VivadoComposer> INFO] Vivado starting run 0x965e5fcc[vexriscv_pe x 1] with axi4mm@aws[F=350.000]: show progress with `vivado_progress /scratch/mio/tapasco_f1/DSE_2
> 019-06-21T22-44-00.422707/bd/axi4mm/aws/vexriscv_pe/001/350.0/axi4mm-aws--vexriscv_pe_1--350.0.log`

As maxTasks was set to 3, runs for 355 and 350 Mhz were started, which were both successful (WNS > 0). At this point, Tapasco continued with even lower frequencies, down to 300 MHz. All of them succeeded. But why didn't Tapasco finish after the first two runs were good and the maximum frequency was found?

m-ober, 2019-06-25:
Actually, I would rather expect Tapasco to try higher frequencies, even if the core.json says 355 MHz is the limit (which was found by OOC evaluation). Manual runs for this core were successful for frequencies as high as 440 MHz.

sommerlukas, 2019-06-25:
I will have to investigate the first problem further. Either the batch should only contain as many elements as `maxTasks` allows or not issue another batch.

The frequency found during OOC synthesis will be used as upper bound for the frequency to limit the design space. You should be able to manually override this start frequency by specifying a frequency just as you would do for a `compose` job (e.g., `@ 400 MHz`).

sommerlukas, 2019-06-26:
Ok, I've investigated this issue further: TaPaSCo DSE groups the elements of the design-space into batches. Only after a complete batch has finished, TaPaSCo will prune elements from the design-space (e.g. because there were successful elements with higher frequency as in your case). 

The size of the batches can be set with the `batchSize` CLI flag. If the flag is not set, the batch size defaulted to the number of processors in the system. That was why TaPaSCo would still run the remaining 10 elements from the batch, even if the first two were successful.

In 6fb76acff5bd2479e7e913fc86599a0bb6267fa7, I've changed the default behavior as follows:
* If the `batchSize`-flag is given, the given value will be used for the DSE batch size
* In case `batchSize` is not given, but `maxTasks` is given, the batch size will be the given maximum number of tasks minus one. 
* In case neither of the two flags is given, the batch size will be the number of processors in the system minus one.

lukasmweber, 2019-06-26:
This issue was closed by #96.

closed 2019-06-26 by lukasmweber