esa-tu-darmstadt/tapasco #100: Raise/remove upper limit of 450 MHz for DSE

m-ober, 2019-06-28:
Starting a DSE:

tapasco --verbose --maxThreads 2 --maxTasks 2 --kernelDir toolflow/examples/knel-examples explore [counter x 1] @ 500 MHz in frequency --platforms pynq

leads to the following batch:

[12:18:52 <pool-1-thread-1: ConcreteBatch> INFO] starting [[counter x 1]] [F=450.000] for axi4mm@pynq

Setting frequencies below 450 MHz works as expected. I also noticed this behaviour when editing the `<AchievedClockPeriod>` entry in the report XML.

I don't know why this limit was chosen, but it seems a bit too low (maybe not for PYNQ but for US+ devices). Also, there is no warning or message generated that notifies about this behaviour. `tapasco help explore` also does not explain such a limit exists and where it is.

sommerlukas, 2019-06-28:
Can you reproduce this issue for other platforms (e.g. Virtex US+)? In TaPaSCo, each platform/target has a range of feasible frequencies and it is very likely that Pynq does not allow such high frequencies for DSE, because they're just not feasible.

m-ober, 2019-06-28:
It's the same for `vcu118`. It looks like this limit comes from here and is hardcoded? 

https://github.com/esa-tu-darmstadt/tapasco/blob/2f31bec7ea89a597b3aac9a86a46c2e8b7fc3b1a/src/main/scala/tapasco/base/json/package.scala#L336

sommerlukas, 2019-06-28:
No, it's not hardcoded. If the field "SupportedFrequencies" in the `platform.json` gives a list of integers, this list will be used as the list of feasible frequencies. Only if this field is not given (which seems to be the case for the `platform.json`s right now), this defaults to a list created by iterating from 50 to 450 in steps of 5. 

So, for the moment you can just specify that field in the `platform.json` to allow for higher frequencies than 450 MHz for your target platform. In the long run, we should fill that JSON-field with a list of sensible values for each `platform.json`.

closed 2019-10-23 by lukasmweber