esa-tu-darmstadt/tapasco #101: Timing failure in DSE is treated as "OtherError"

m-ober, 2019-06-28:
According to `tapasco help explore`, there are three types of errors: Placer Error, Timing Failure and Other Error, where Timing Failure should _"generate a feedback element: A new design space element is generated for the same Composition, but with a lower target frequency. The frequency is computed from the 'worst negative slack' reported by the composer tools"_.

But a timing failure seems to result in an "OtherError":

[12:50:37 <pool-1-thread-2: VivadoComposer> ERROR] Routing failed after 0:00:39 (Total Elapsed: 0:06:33)                                                                                       
[12:50:37 <pool-1-thread-2: VivadoComposer> ERROR] Vivado finished with non-zero exit code: 1 for 0xa0c3925e[counter x 1] with axi4mm@pynq[F=445.000] in '/home/mober/projects/tapasco/DSE_2019
-06-28T12-37-28.785/bd/axi4mm/pynq/counter/001/445.0'                                                                                                                                          
[12:50:37 <pool-1-thread-2: ComposeTask> INFO] compose run [counter x 1]@445.000 MHz for axi4mm@pynq finished, **result: OtherError**, bitstream file: '', logfile: '/home/mober/projects/tapasco/D
SE_2019-06-28T12-37-28.785/bd/axi4mm/pynq/counter/001/445.0/axi4mm-pynq--counter_1--445.0.log', utilization report: '', timing report: '' 

Looking in the logfile, it was clearly a timing failure:

CRITICAL WARNING: [Timing 38-282] The design failed to meet the timing requirements. Please see the timing summary report for details on the timing violations.
timing failure, WNS: -2.521
    while executing
"error "timing failure, WNS: $wns""
    (procedure "platform::generate" line 84)
    invoked from within
"platform::generate"
    (file "/home/mober/projects/tapasco/DSE_2019-06-28T12-37-28.785/bd/axi4mm/pynq/counter/001/445.0/pynq.tcl" line 136)

If this is the intended behaviour, the documentation should be updated (remove the "Timing Failure" error from the list/mention a timing failure will also be shown as "Other Error").

sommerlukas, 2019-06-28:
Did TaPaSCo in your case continue the DSE with the next lower frequency, despite the `OtherError`?

m-ober, 2019-06-28:
Yes, but using the smallest possible step (5 MHz). According to the help text, I would expect the frequency to be lowered by several hundred MHz in the example above.

sommerlukas, 2019-06-28:
I fixed the detection of `TimingError` in e087fc8bb2f783b273b260af1431060de1b020dc, it now correctly detects timing errors if Vivado gives a non-zero exit code:
> Vivado finished, but did not achieve timing closure for 0x482fd4dc[arraysum x 1] with axi4mm@pynq[F=450.000], WNS: -2.768, max delay path: Some(TimingPath(system_i/arch/out_0/bic/s00_couplers/auto_us/inst/gen_upsizer.gen_full_upsizer.axi_upsizer_inst/USE_READ.read_addr_inst/GEN_CMD_QUEUE.cmd_queue/USE_FF_OUT.USE_RTL_OUTPUT_PIPELINE.M_MESG_Q_reg[11]/C,system_i/arch/out_0/bic/s00_couplers/auto_us/inst/gen_upsizer.gen_full_upsizer.axi_upsizer_inst/USE_READ.gen_non_fifo_r_upsizer.read_data_inst/USE_RTL_LENGTH.first_mi_word_q_reg/CE,-2.768)) in '/tmp/DSE_2019-06-28T15-50-15.363059/bd/axi4mm/pynq/arraysum/001/450.0'

Regarding the step width in the DSE resulting from a timing failure: The behavior of the DSE was changed in the past to not jump over frequency elements, because the WNS does not provide reliable information. The continuation of the DSE with only 5 MHz less is the intended behavior. I will update the documentation in `tapasco --help explore` accordingly.

sommerlukas, 2019-06-28:
Closed by #102;

closed 2019-06-28 by sommerlukas