esa-tu-darmstadt/tapasco #106: --verbose log output is (sometimes) doubled

m-ober, 2019-07-04:
Can't say (yet) when exactly it happens, but sometimes, when using `--verbose` mode to watch the logfile, the lines are output twice, like here:
                                                                                                                                                
> [23:35:29 <Thread-6: VivadoComposer> INFO] *** Running vivado                                                                                                                                  
> [23:35:29 <Thread-6: VivadoComposer> INFO]     with args -log cl_top.vds -m64 -product Vivado -mode batch -messageDb vivado.pb -notrace -source cl_top.tcl                                     
> [23:35:29 <Thread-6: VivadoComposer> INFO]                                                                                                                                                     
> [23:35:33 <Thread-6: VivadoComposer> INFO]                                                                                                                                                     
> [23:35:33 <Thread-6: VivadoComposer> INFO] *** Running vivado                                                                                                                                  
> [23:35:33 <Thread-6: VivadoComposer> INFO]     with args -log cl_top.vds -m64 -product Vivado -mode batch -messageDb vivado.pb -notrace -source cl_top.tcl                                     
> [23:35:33 <Thread-6: VivadoComposer> INFO]                                                                                                                                                     
> [23:35:35 <Thread-6: VivadoComposer> INFO]                                                                                                                                                     
> [23:35:35 <Thread-6: VivadoComposer> INFO] ****** Vivado v2018.3 (64-bit)                                                                                                                      
> [23:35:35 <Thread-6: VivadoComposer> INFO]   **** SW Build 2405991 on Thu Dec  6 23:36:41 MST 2018                                                                                             
> [23:35:35 <Thread-6: VivadoComposer> INFO]   **** IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018                                                                                             
> [23:35:35 <Thread-6: VivadoComposer> INFO]     ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.                                                                                        
> [23:35:35 <Thread-6: VivadoComposer> INFO]                                                                                                                                                     
> [23:35:35 <Thread-6: VivadoComposer> INFO] Sourcing tcl script '/scratch/mio/tapasco_f1/toolflow/TCL/common/scripts/Vivado_init.tcl'                                                           
> [23:35:35 <Thread-6: VivadoComposer> INFO] 0 Beta devices matching pattern found, 0 enabled.                                                                                                   
> [23:35:37 <Thread-6: VivadoComposer> INFO] source cl_top.tcl -notrace                                                                                                                          
> [23:35:39 <Thread-6: VivadoComposer> INFO]                                                                                                                                                     
> [23:35:39 <Thread-6: VivadoComposer> INFO] ****** Vivado v2018.3 (64-bit)                                                                                                                      
> [23:35:39 <Thread-6: VivadoComposer> INFO]   **** SW Build 2405991 on Thu Dec  6 23:36:41 MST 2018                                                                                             
> [23:35:39 <Thread-6: VivadoComposer> INFO]   **** IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018                                                                                             
> [23:35:39 <Thread-6: VivadoComposer> INFO]     ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.                                                                                        
> [23:35:39 <Thread-6: VivadoComposer> INFO]                                                                                                                                                     
> [23:35:39 <Thread-6: VivadoComposer> INFO] Sourcing tcl script '/scratch/mio/tapasco_f1/toolflow/TCL/common/scripts/Vivado_init.tcl'                                                           
> [23:35:39 <Thread-6: VivadoComposer> INFO] 0 Beta devices matching pattern found, 0 enabled.                                                                                                   
> [23:35:39 <Thread-6: VivadoComposer> INFO] source cl_top.tcl -notrace                                                                                                                          
> [23:35:43 <Thread-6: VivadoComposer> INFO] Command: synth_design -top cl_top -part xcvu9p-flgb2104-2-i -retiming -mode out_of_context -max_uram_cascade_height 1                               
> [23:35:43 <Thread-6: VivadoComposer> INFO] Starting synth_design                                                                                                                               
> [23:35:43 <Thread-6: VivadoComposer> INFO] Attempting to get a license for feature 'Synthesis' and/or device 'xcvu9p'

sommerlukas, 2019-07-05:
Interestingly, the doubled output has different timestamps, e.g. `*** Running vivado` is reported twice with 4 seconds difference in the timestamp. 
@lukasmweber has touched upon some related code for the status information feature, maybe he has an idea what is going on.

lukasmweber, 2019-07-08:
Are there any new insights on when exactly this behavior occurs? If not I am going to try to reproduce it with the current state of development.

m-ober, 2019-07-08:
I have not yet spend any time looking further into this issue. If you cannot reproduce this, I should note that I'm using OOC synthesis, i.e. I have removed `set_property synth_checkpoint_mode None` from the TCL template and use `-mode out_of_context` as synth setting. Maybe Tapasco is confused if multiple Vivado threads are spawned. Although the number of threads does not correlate with the multiple log output (e.g. I'm using 6 threads but output is doubled)

lukasmweber, 2019-07-09:
I have located the issue and started working on a fix.

lukasmweber, 2019-07-09:
Providing a fix for this is rather complex due to the way verbose output is currently handled. To untangle the corresponding log-files and provide coherent and synchronized logs would incur significant overhead. In addition, actual logfiles seem to be unaffected by this bug, only the actual console output is broken. Thus I am not going to fix this. This could potentially be fixed in the future when File-Mgmt is reworked.

closed 2019-07-09 by lukasmweber