esa-tu-darmstadt/tapasco #116: 24h Timeout: Document or remove entirely

m-ober, 2019-08-13:
My synthesis got killed after 24 hours.

```
[18:30:37 <Thread-7: InterruptibleProcess> WARN] thread interrupted, destroying external process
[18:30:37 <pool-1-thread-1: VivadoComposer> ERROR] Writing Bitstream failed after 0:34:22 (Total Elapsed: 24:00:00)
[18:30:37 <pool-1-thread-1: VivadoComposer> ERROR] Vivado timeout for 0x411b37fc[BAUDIO_4000-FP x 3] with axi4mm@aws[F=125.000] 
```

Don't know why this was introduced, but it should at least be documented in `tapasco help compose`, set to a more sensible value or removed entirely. Better to manually kill a stuck process instead of wasting 24 hours of compute time :grinning: 

https://github.com/esa-tu-darmstadt/tapasco/blob/619460939c46040eb16a3cec1cccb82f51a3cffa/src/main/scala/tapasco/activity/composers/VivadoComposer.scala#L83

jahofmann, 2019-08-13:
Who doesn't like random, undocumented, behavior...

sommerlukas, 2019-10-16:
I think it make sense to have a built-in timeout, in particular for not permanently supervised usage scenarios, e.g., running a large number of composes on a cluster. 

I've increased the timeout to 48 hours on branch `feature/vivado-timeout` and also documented that behavior in the CLI.

@m-ober, @jahofmann: From your experience with Virtex US+ and Alveo devices, is a 48 hour limit sufficient for synthesis & implementation on recent FPGA devices?

sommerlukas, 2019-10-16:
I picked up that idea from @m-ober in b38e3a9c0c6e05faaed5968eef13374c01179561. It is now possible to configure a timeout for implementation runs in the `platform.json` using the field `ImplementationTimeout`.

@m-ober, @jahofmann What do you think?

jahofmann, 2019-10-17:
Yes, that would be the cleanest option. If a Zynq design runs for e.g. four hours we know that it most likely failed, a 48 hour timeout would serve no purpose here.

We can start with sensible defaults for the platforms and improve upon them later on, for instance based on CPU cores available etc.

closed 2019-10-23 by lukasmweber