esa-tu-darmstadt/tapasco #123: List publications about/using TaPaSCo

sommerlukas, 2019-10-14:
New users can learn a lot about TaPaSCo and its features & capabilities from the scientific publications about/using TaPaSCo. 

I've started a page in the [wiki](https://github.com/esa-tu-darmstadt/tapasco/wiki/Publications) that lists all these publications. @jahofmann, @cahz please check the page for correctness and completeness. 

We should probably also have a link to these publications in the README, either as part of #121 or as a separate hotfix.

lukasmweber, 2019-10-23:
A corresponding Link and Description is now in the Readme of the branch for #121.

sommerlukas, 2020-05-07:
I updated the publications for release 2020.4;

sommerlukas, 2020-10-13:
No new publications using TaPaSCo since 2020.4 for 2020.10 (some currently in the pipeline), moving this permanent issue to the next milestone.