esa-tu-darmstadt/tapasco #131: [CI] Use valid Java versions in SDKMAN!

cahz, 2019-10-17:
Version updates in SDKMAN! often break our CI. Maybe we should use the list of available versions and then automatically select an available version.

sommerlukas, 2019-10-17:
Maybe this is a good point to discuss if we want to use SDKMAN by default in the future. As we no longer depend on up-to-date versions of ``sbt`, we could also use the OS-provided Java installation.

IMHO there's not much reason to use SDKMAN anymore, in particular if it breaks our builds on a regular basis. 

Instead, we could test with the Java versions delivered with relevant Linux distributions, just as we do for the kernel module and the runtime libraries.

@lukasmweber, @cahz, @jahofmann: What do you think?

sommerlukas, 2019-10-17:
This would also allow us to make `java` a required dependency of the OS-package built in #40.

jahofmann, 2019-10-17:
+1, There doesn't seem any reason to keep it and it makes the installation process unnecessary clunky.

commit 5df79e6c (2019-11-05): Removes SDKMAN and uses default Java versions

    - Fixes #131
closed 2019-11-05 by jahofmann