esa-tu-darmstadt/tapasco #133: Provide flag to skip synthesis

cahz, 2019-10-19:
For testing purposes, it could be useful to have a command line argument to skip synthesis and bitstream generation. This would reduce runtime so that we can increase test coverage of all platforms in the CI.

This could prevent broken system compositions for rarely used platforms (e.g. bugs like e7d1a2fa94d277f45863cd817cd99bd51af67c66)

mhrtmnn, 2020-06-06:
Seeing that there already is the `--dryRun` flag, for both HLS and Compose jobs, how would this feature differ from a dry run?

cahz, 2020-06-06:
Do you mean the `--debugMode` flag for compose? The difference there is that vivado is not even run. The idea of this issue is to execute all steps like system composition in vivado, but being able to skip the time consuming tasks of synthesis and implementation.

sommerlukas, 2020-06-08:
@cahz: Can point @mhrtmnn to the exact point (e.g. in TCL) where you want to stop the process?

cahz, 2020-06-08:
After generating the verilog wrapper should be a good point: https://github.com/esa-tu-darmstadt/tapasco/blob/develop/toolflow/vivado/common/design.master.tcl.template#L113

mhrtmnn, 2020-06-08:
OK I see, thanks!

closed 2020-06-16 by lukasmweber

dantor, 2020-06-18:
Thanks, super useful!

PR #215 (2020-06-16): Feature/add flag to skip synthesis
Fixes #133