esa-tu-darmstadt/tapasco #141: Rebuild the stack of benchmarks

jahofmann, 2019-11-04:
As part of the work in issue #79, the benchmarks that have been deprecated for a while have been removed as they were not functional anymore.

To evaluate the performance of TaPaSCo properly, again, there should be a replacement that checks the different performance critical parts of TaPaSCo.

There were four different tests:
 - benchmark-alloc-free: Check memory allocation performance
 - benchmark-cd: ???
 - benchmark-latency: Counter execution latency
 - benchmark-mem: Bandwidth for data movements -> Replaced by bandwidth tool

Additional tests should be added for the different parts of the architecture.

cahz, 2020-10-07:
@jahofmann is this still relevant?

jahofmann, 2020-10-07:
Sure, would still be nice to have more such tools to better catch regressions.