esa-tu-darmstadt/tapasco #150: Prepare Release 2019.10

jahofmann, 2019-11-11:
- [ ] Finish #121 
- [ ] Finish #90 
- [ ] Finish #80 
- [ ] Finish #47 
- [ ] Update version specifiers
- [ ] Test Pynq Bitstream Generation
- [ ] Test Zedboard Bitstream Generation
- [ ] Test VC709 Bitstream Generation
- [ ] Test VC709 Default Tools (memcheck, benchmark, debug, arrayinit, arraysum, arrayupdate)
- [ ] Test ZC706 Bitstream Generation
- [ ] Test ZC706 Default Tools (memcheck, benchmark, debug, arrayinit, arraysum, arrayupdate)
- [ ] Test NetFPGA SUME Bitstream Generation
- [ ] Test NetFPGA SUME Default Tools (memcheck, benchmark, debug, arrayinit, arraysum, arrayupdate)
- [ ] Test AU250 Bitstream Generation
- [ ] Test ZCU102 Bitstream Generation

jahofmann, 2019-11-11:
@cahz and @lukasmweber How long do you need to complete the assigned issues? Need any help?

jahofmann, 2019-11-11:
I did the ZC706 test already and will test the NetFPGA SUME tomorrow.

jahofmann, 2019-11-13:
Release Branch created

lukasmweber, 2019-11-13:
I will start with testing Pynq and Zedboard Bitstream generation.

jahofmann, 2019-11-18:
VC709 Benchmark added

Renamed to .log because gitlab doesn't like other file types...

[mountdoom_vc709.benchmark.log](https://github.com/esa-tu-darmstadt/tapasco/files/3858085/mountdoom_vc709.benchmark.log)

cahz, 2019-11-18:
ZC706 Benchmark

[zc706.benchmark.log](https://github.com/esa-tu-darmstadt/tapasco/files/3858121/zc706.benchmark.log)

jahofmann, 2019-11-18:
Plotted the benchmarks and added a script for the future (Improvements needed)

[performance.pdf](https://github.com/esa-tu-darmstadt/tapasco/files/3858462/performance.pdf)

jahofmann, 2019-11-18:
For the next time we have to properly specify what kind of bitstreams we're using. Otherwise the benchmarks aren't really comparable.

For my benchmarks I've been using 32 Counter @ 250MHz on a Ryzen 1600x.

cahz, 2019-11-18:
As discussed last week, I used a bitstream with two counters on the ZC706.

jahofmann, 2019-11-19:
Release Notes
--------------------
Any input @lukasmweber @cahz @sommerlukas?

------------

We are happy to announce TaPaSCo 2019.10 coming right to you with a number of structural changes.

Directory Structure
===============
The directory structure has been changed completely. The individual parts of TaPaSCo are now separated into the Toolflow and Runtime directories to avoid confusion.

TaPaSCo Setup
============
The novel `tapasco-init` script enables out-of-tree builds to simplify using multiple TaPaSCo builds in parallel.

Toolflow Changes
=======================
The toolflow is now built by Gradle instead of SBT. Along with these changes, we now provide OS packages for leading Linux distributions. Compilation of TaPaSCo is now longer necessary for a majority of users.

Quality of Life
===========
A number of quality of life changes targets the usability of TaPaSCo. For instance, the Readme is updated and should be much clearer.

TaPaSCo Status Core
=================
The status core is now based on Protobuf instead of using a custom format. The new format is much more flexible and provides much more indepth information about the bitstream.

Bugfixes
=======
In total this release closes 34 [Issues](https://github.com/esa-tu-darmstadt/tapasco/issues?q=is%3Aissue+milestone%3A2019.10+is%3Aclosed) in over 270 commits.

jahofmann, 2019-11-19:
Results for NetFPGA Sume on i5-6600K
There is some issue with the latency measurements (negative measurements)
[netfpga_sume.benchmark.log](https://github.com/esa-tu-darmstadt/tapasco/files/3863093/netfpga_sume.benchmark.log)
 which I have to fix before continuing.

closed 2019-11-19 by jahofmann