esa-tu-darmstadt/tapasco #155: Support --skipEvaluation parameter for compose task

m-ober, 2019-11-16:
I would like to use Tapasco like this:

```
                                 vvvvvvvvvvvvvvvv
tapasco --verbose --maxThreads 2 --skipEvaluation \
--kernelDir ../toolflow/examples/kernel-examples/ \
compose [arrayinit x 128] @ 100 MHz --platforms aws --deleteProjects false

```

So I don't need to import the kernel first (where  `--skipEvaluation` is supported).

jahofmann, 2019-11-16:
I had the same wish some time ago, but it was rejected for some reason. Can you remind me what it was @lukasmweber?

lukasmweber, 2019-11-17:
I am not sure if i remember this correctly, but i think that the main problem here is, that `--skipEvaluation` is a contextual flag that only relates to `import` and `hls` Jobs. At the moment, TaPaSCo accepts two kinds of flags, ones that are global and others that are specific to the Job (eg. `tapasco <global flags> job <specific flags>`. Adding `--skipEvaluation` as a global flag is not really semantically sound, since it is a flag that is non-global and can only apply to two kinds of Jobs.

I am guessing that the reason for rejecting this previously (@jahofmann ) was probably, that we are intending to rework the CLI interface as a whole in one of the forthcoming releases, where one of the potential goals could be the removal of this separation of global and specific flags.
At this time, it would be possible to encode the corresponding jobs (eg a `hls`-job with `--skipEvaluation` and a trailing `compose`-job) using are `job.json` File.

sommerlukas, 2019-11-17:
IIRC, the (my) reasoning was a different one: We already have a `import`-command that supports all these flags and implements the corresponding behavior. In addition to that, we have the `bulkimport` which allows to import multiple cores with a single command. 

Using `import` before `compose` (or `dse`) is also the way we teach the TaPaSCo design flow to new users, e.g., in the tutorial. 

I personally would rather get rid of the automatic import in the `compose`-command than implement the `--skipEvaluation` flag for it. With the `--skipEvaluation`-flag, the `import`-command takes only seconds to complete, so IMHO there is no need to save users from running that command before `compose`. I also think that this "automagic" import can be confusing for inexperienced users and lead to cores being stored in undesired locations.

jahofmann, 2019-11-17:
The question then is: why make the process less comfortable when the same result can be achieved by running just one command?

I don't think we gain anything from removing automatic import. The goal of every decision here should be to make it as user friendly as possible.

Back then my preferred solution would have been to make skipEvaluation the default and run the evaluation only if DSE is desired. All in the spirit of "make the common case fast" and make the whole process as much fun and seamless as possible.

m-ober, 2019-11-17:
@sommerlukas  I see your point about running `import` before `compose` is just one command more - but on the other hand, it is one more command to type :wink: 

@lukasmweber When `compose` is running an `import`/`hls` task (as prerequisite), it should also support the flags supported by those tasks. At least that is my expectation of how the user interface should work. If a command I type results in two tasks being run (`hls`, then `compose`), why can I only give contextual flags for one of the two tasks?

What's wrong with accepting `--skipEvaluation` even if no `import`/`hls` task is executed? This flag skips evaluation if there is something to be imported. If nothing needs to be imported/synthesized, evaluation is also skipped because there is nothing evaluate.

What I totally aggree with is that having "global" and "contextual" flags is incredibly confusing. "Is this flag global or task specific?" - "How is that flag unknown, I just used it in the previous command ... oh, it's not supported for this task"...

sommerlukas, 2019-11-17:
The distinction between global and contextual flags is a design decision that is surely debatable, but I do see why the distinction was made, basically there are two options:
* Supporting some flags only on some commands, leading to errors if a flag was specified on a command that does not support that flag (this is how it is currently implemented).
* Parsing all flags on all commands, but giving a warning if flags are ignorred.

In the first case, the user gets very clear feedback (something we have been working on very much in the last few months) in the form of an error if he attempts to do something not supported. In the latter case, he might wonder why his desired behavior was not realized by TaPaSCo (e.g. skipping evaluation on a `compose`-command) and might need to identify the corresponding warning in a potentially very long output. For this reason, IMHO it is not a bad decision to support some flags only on some commands and giving errors if the flags is used on a wrong command.

m-ober, 2019-11-17:
> In the first case, the user gets very clear feedback

I would like to disagree :wink:   _"invalid arguments: expected End at 150"_ is everything but clear. Doesn't this flag exist at all (because of a typo, maybe), am I using a global flag as contextual? Or the other way round? Am I using it correctly as contextual flag but it's not supported for this task? So this message leaves a lot of questions open...

> might need to identify the corresponding warning in a potentially very long output

I would accept this argument for flags which are supposed to _do something_, e.g. set the number of threads, set a strategy etc. - I don't want those kind of flags to be "silently" ignored, if they are not supported in the current context.  This could lead to unexpected results if I'm not carefully watching the logs. But in the case of the `--skipEvaluation` flag, it's a _"don't do something"_. If _something_ is not supported in the current context, not doing it is very easy and the presence of this (in this case) unsupported flag does not change the result at all.

sommerlukas, 2019-11-17:
> So this message leaves a lot of questions open..

I agree, this message has to be improved. But this has nothing to do with the fact, that there are contextual flags, this is just a problem of the hand-crafted CLI-parser.

> But in the case of the --skipEvaluation flag, it's a "don't do something". 

I disagree , this flags **does** something: It skips the evaluation of cores. The unexpected result in this case is a potentially much higher runtime for the command, which can even cost you money/core-hours if you run the command in an environment that charges you for compute time. 

Any flags passed to any TaPaSCo command changes the expected behavior, otherwise there was no need to specify it.

jahofmann, 2019-11-17:
Maybe we should separate this discussion into the corresponding rework CLI issue.

The question here is, if skipEvaluation should become the default and if it should be a parameter of the compose command.

sommerlukas, 2019-11-17:
Yes, you are wright. Even if we decided to remove the distinction between global and contextual flags, this could only happen as part of a major overhaul of the CLI.

Regarding the original issue, here is what I think are potential options and the problems they bear: 

* Making `--skipEvaluation` the default behavior for automatic imports as part of a `compose`-command. This is problematic, because if the user wants to use the same core for DSE later on, no evaluation information will be available and he has to manually delete the core from the `coreDir` to be able to re-import it with evaluation. 

* Adding the flag to the `compose`-command: As stated above, the import and evaluation has nothing to do with the `compose`-command itself, so it can be confusing to have that flag here. 

* Not adding the flag to `compose` and require the user to do a explicit `import` if he wants to skip evaluation. The command will finish almost instantly.

Considering all three options and the fact that what we "officially" teach is closest to the last option, I personally would prefer to go with option 3 for now.

jahofmann, 2019-11-17:
Please read what I wrote above.

We should not adhere to some "arcane" teaching, but make TaPaSCo as easy to use as possible. I see the import as part of the compose command and accordingly the skipEvaluation flag makes perfect sense.

Your first point is only a problem because DSE behaves that way. It doesn't make sense, in the general case, that Import runs the Evaluation. That should be the task of DSE. If it doesn't find the evaluation result it can simply run the evaluation on-demand. No manual intervention necessary. The tool behaves as desired. In addition, no unnecessary import runs or evaluations needed.

In my opinion running import should be the exception, not the rule. What does the import run add to the user experience, except for one additional command?

sommerlukas, 2019-11-17:
I would like to keep the ˋimportˋ-command, because it allows users to import cores into the ˋcoreˋ-directory without having to know the directory structure. Besides that, ˋimportˋ with evaluation allows to use ˋcoreStatsˋ for an early assessment of a core's performance and resources. 

But if you consider ˋcomposeˋ without prior evaluation to be the most common case, I would propose to go one step further and do the following:
* Make ˋskipEvaluationˋ the default for all commands executing an import, except for ˋdseˋ.
* Keep the automatic import on ˋcomposeˋ. 
* Add a flag ˋevaluateˋ to the ˋimportˋ-command which executes the import. 
* If a core that has not yet been evaluated is used in DSE, ˋdseˋ will automatically run evaluation. The information stored in the ˋcoreDirˋ should be sufficient. 

tl;dr: Make ˋskipEvaluationˋ default, replace with ˋevaluateˋ-flag on ˋimportˋ, automatically run evaluation as part of DSE if required.

@jahofmann, @m-ober: What do you think?

m-ober, 2019-11-18:
Sounds good to me.

PR #331 (2022-10-19): Automize Evaluation
Closes #155;

With this PR, `--skipEvaluation` is officially removed from TaPaSCo. Instead the hls and import subcommands automatically skip the Evaluation Step. If Evaluation is required by the user `--evaluate` can be used. If DSE is triggered (which requires Evaluation, the cores without Evaluation Results will automatically be evaluated).
closed 2022-10-21 by lukasmweber