esa-tu-darmstadt/tapasco #163: SegFault when working with multiple parallel jobs

jojowi, 2020-02-22:
Start multiple Jobs in parallel using the C++ API and store job_futures and RetVals in std::vectors:

```
std::vector<tapasco::job_future> jobs;
std::vector<tapasco::RetVal<unsigned long>> retvals;
unsigned long rets[instances];
for (int instance = 0; instance < instances; instance++) {
        rets[instance] = 0;
        retvals.push_back(tapasco::makeRetVal(rets[instance]));
        auto job = tapasco.launch(PE_ID, retvals[instance], ...);
        jobs.push_back(job);
}
```

Waiting on the completion of the first job `jobs[0]();` crashes with a Segmentation Fault; waiting on the other jobs works fine.

This can be fixed by reserving enough space in the vectors directly after their creation:

```
std::vector<tapasco::job_future> jobs;
jobs.reserve(instances);
std::vector<tapasco::RetVal<unsigned long>> retvals;
retvals.reserve(instances);
```

The Segmentation Fault does not really give a hint how to solve the problem, maybe there is a way to catch this more user-friendly?

jahofmann, 2020-04-17:
@jojowi is reserving the jobs vector really necessary? From looks I would guess that this is more a problem with the return value as the pointer to this memory is stored and accessed after completion.

jojowi, 2020-04-20:
I just tested this, you're right. It is sufficient to reserve the required space in the return value vector.

While testing this I also noticed something else:
I create a RetVal and push it to the vector:
```
tapasco::RetVal<unsigned long> ret_val(rets[instance]);
retvals.push_back(ret_val);
```
If I take the RetVal from the vector when lauching the job
`tapasco.launch(PE_ID, retvals[instance], ...);`
there is a Segmentation Fault unless I reserve the required space in the return value vector as explained above.

But if use the ret_val variable instead
`tapasco.launch(PE_ID, ret_val, ...);`
it works, even without reserving space in any of the vectors.

jahofmann, 2020-04-29:
I am not sure we can fix this with the way C++ works. In Rust these kinds of errors at least can not happen.

We simply transfer a pointer and if you change the memory it is pointing to while the runtime is working on it, it is bound to crash.

Anyway, will move this issue to another Milestone as I am not sure we can even fix it with the way C++ is designed.

wirthjohannes, 2021-04-23:
No longer relevant due to new runtime