esa-tu-darmstadt/tapasco #168: TaPaSCo fails successfully

jahofmann, 2020-04-03:
It seems like error codes are not properly forwarded, which can be quite confusing:
```
[16:25:30 <main: Tapasco$> INFO] Running with configuration: [Configuration @/home/wimi/jah/projects/tapasco/tapasco_hoplite/default.cfg]
Verbose = None
KernelDir = /home/wimi/jah/projects/tapasco/tapasco_hoplite_build/kernel
CoreDir = /home/wimi/jah/projects/tapasco/tapasco_hoplite_build/core
ArchDir = /home/wimi/jah/projects/tapasco/tapasco_hoplite/toolflow/vivado/arch
PlatformDir = /home/wimi/jah/projects/tapasco/tapasco_hoplite/toolflow/vivado/platform
Slurm = false
Parallel = false
MaxThreads = unlimited
MaxTasks = unlimited
Jobs = List(ComposeJob([counter x 1],122.0,Vivado,None,Some(ArrayBuffer(vc709)),Some(ArrayBuffer(Feature(NOC,FMap(Map(enabled -> FString(true)))), Feature(Debug,FMap(Map(interfaces -> FString({ /noc/ic_s_02_02/M00_AXI /noc/clk_wiz/clk_out1 /noc/clk_wiz/locked }), enabled -> FString(true)))))),None,None,Some(false)))
[16:25:31 <main: Compose$> INFO] all HLS tasks finished successfully, beginning compose run...
[16:25:31 <pool-1-thread-1: VivadoComposer> INFO] Vivado starting run 0xa0c3925e[counter x 1] with axi4mm@vc709[F=122.000]: show progress with `vivado_progress /home/wimi/jah/projects/tapasco/tapasco_hoplite_build/compose/axi4mm/vc709/counter/001/122.0+Debug+NOC/axi4mm-vc709--counter_1--122.0.log`
[16:25:33 <Thread-6: VivadoComposer> INFO] Started System Composition (Total Elapsed: 0:00:02)
[16:25:39 <pool-1-thread-1: VivadoComposer> ERROR] System Composition failed after 0:00:06 (Total Elapsed: 0:00:08)
[16:25:39 <pool-1-thread-1: VivadoComposer> INFO] Vivado finished successfully for 0xa0c3925e[counter x 1] with axi4mm@vc709[F=122.000], WNS: 0.003, bitstream file is here: '/home/wimi/jah/projects/tapasco/tapasco_hoplite_build/compose/axi4mm/vc709/counter/001/122.0+Debug+NOC/axi4mm-vc709--counter_1--122.0.bit'
[16:25:39 <pool-1-thread-1: ComposeTask> INFO] compose run [counter x 1]@122.000 MHz for axi4mm@vc709 finished, result: Success, bitstream file: '/home/wimi/jah/projects/tapasco/tapasco_hoplite_build/compose/axi4mm/vc709/counter/001/122.0+Debug+NOC/axi4mm-vc709--counter_1--122.0.bit', logfile: '/home/wimi/jah/projects/tapasco/tapasco_hoplite_build/compose/axi4mm/vc709/counter/001/122.0+Debug+NOC/axi4mm-vc709--counter_1--122.0.log', utilization report: '/home/wimi/jah/projects/tapasco/tapasco_hoplite_build/compose/axi4mm/vc709/counter/001/122.0+Debug+NOC/utilization.txt', timing report: '/home/wimi/jah/projects/tapasco/tapasco_hoplite_build/compose/axi4mm/vc709/counter/001/122.0+Debug+NOC/timing.txt'
[16:25:39 <main: Compose$> INFO] all Compose tasks finished
[16:25:40 <main: Tapasco$> INFO] TaPaSCo finished successfully
```

sommerlukas, 2020-04-03:
Did you check for the return code of `vivado`? Is the forwarding broken in the TCL script or on the interface between TaPaSCo and Vivado?

jahofmann, 2020-04-03:
The Log already says `[16:25:39 <pool-1-thread-1: VivadoComposer> ERROR] System Composition failed after 0:00:06`, so that information should be available?

sommerlukas, 2020-04-03:
But it also says `Vivado finished successfully for 0xa0c3925e[counter x 1] with axi4mm@vc709[F=122.000], WNS: 0.003, bitstream file is here: '/home/wimi/jah/projects/tapasco/tapasco_hoplite_build/compose/axi4mm/vc709/counter/001/122.0+Debug+NOC/axi4mm-vc709--counter_1--122.0.bit'`, so I'm not sure if the Vivado return code actually signals failure.

Could you please check that?

jahofmann, 2020-04-03:
```bash
Creating component msix_intr_ctrl ...
  VLNV: esa.informatik.tu-darmstadt.de:user:MSIXIntrCtrl:1.0
Unused Interrupts: 2, 3 are tied to 0
Creating xlconstant irq_unused with 1-bit width and value 0 ...
  VLNV: xilinx.com:ip:xlconstant:1.1
Creating xlconcat interrupt_concat_design with 4 ...
  VLNV: xilinx.com:ip:xlconcat:2.1
Subsystem intc complete.
Creating subsystem memory ...
invalid command name "KEWORKWEGIOWSERGOWREIMGIW"
    while executing
"KEWORKWEGIOWSERGOWREIMGIW"
    (procedure "create_mig_core" line 2)
    invoked from within
"create_mig_core "mig""
    (procedure "create_subsystem_memory" line 25)
    invoked from within
"create_subsystem_memory"
    ("eval" body line 1)
    invoked from within
"eval $cmd"
    ("foreach" body line 9)
    invoked from within
"foreach ss $sss {
      set name [string trim $ss "/"]
      set cmd  "create_subsystem_$name"
      puts "Creating subsystem $name ..."
      if {[ll..."
    (procedure "platform::create" line 12)
    invoked from within
"platform::create"
    (file "vc709.tcl" line 113)
 jah@turgon  
>  echo $?
1
````

Return value is a 1

sommerlukas, 2020-04-03:
Ok, so the problem is between Vivado and TaPaSCo. Can you provide a simple reproducer?

jahofmann, 2020-04-03:
Add random characters in a TCL script, e.g.
```diff
diff --git a/toolflow/vivado/platform/pcie/pcie_base.tcl b/toolflow/vivado/platform/pcie/pcie_base.tcl
index 7ade275..201ecb5 100644
--- a/toolflow/vivado/platform/pcie/pcie_base.tcl
+++ b/toolflow/vivado/platform/pcie/pcie_base.tcl
@@ -109,6 +109,7 @@
   # connected to the PCIe bridge (required ports external).
   # @param irqs List of the interrupts from the threadpool.
   proc create_subsystem_intc {} {
+    THISISACRASH
     set irqs [arch::get_irqs]
     puts "Connecting [llength $irqs] interrupts .."
     # create hierarchical ports
```

and run TaPaSCo e.g. with
```bash
tapasco compose '[counter x 1]' @ 123MHz -p vc709
```

sommerlukas, 2020-04-30:
Fixed in #172;

PR #172 (2020-04-30): Correctly detect Vivado failure
Even if Vivado fails with a non-zero exit code, TaPaSCo still assumes the run to be successful, if a timing report is found to make up for cases in which Vivado fails after all essential steps of the `compose` process have been finished. 

However, if the same composition was used before, the outdated timing report is detected, causing TaPaSCo to report the wrong return code, see #168. 

This fixes #168 by deleting the outdated timing report.
closed 2020-04-30 by sommerlukas