esa-tu-darmstadt/tapasco #178: Status information for ECC memory

jojowi, 2020-04-17:
There is currently no easy way to check status information for ECC memory (e.g. is ECC enabled? did ECC errors occur?).

Besides this an easy way to configure ECC online (disable ECC decoding; reset ECC error counter) would also be nice.

Any comments where a good place for this would be?
My suggestion would be in tapasco-debug, either on the "Monitor blue infrastructure" page or on a new "ECC memory" page.

jahofmann, 2020-04-17:
tapasco-debug sounds good and we can put a fitting flag into the status core.

I'm building supporting infrastructure for things like this in the new runtime. Maybe we should wait till this is ready before changing the status core twice.

jojowi, 2020-04-17:
Ok, then I'll wait for the new runtime