esa-tu-darmstadt/tapasco #179: Tapasco multi device support

jojowi, 2020-04-28:
When there are multiple TaPaSCo devices on a host, it should be possible to select the correct device where necessary (e.g. when using tapasco-load-bitstream).

This will also need a mapping between JTAG and PCIe Slot.

Will be postponed until the new runtime is ready

jojowi, 2020-04-28:
Additional features for tapasco-load-bitstream would also be useful (e.g. list all available devices)

jahofmann, 2020-04-28:
I imagine some functionality such as:

```bash
> tapasco-load-bitstream --list-devices
0 -> Zynq MP
1 -> PCIe JTAG XUPVVH
2 -> PCIe Direct VC709
```

```bash
> tapasco-load-bitstream 0 some-bitstream.bit
Ok
```
The new runtime already works fine with the scenario and devices can be selected as needed and as indicated by TLKM.

What needs to be done:
- [ ] Move PCIe shellscript into Python based tapasco-load-bitstream
- [ ] Use Vivado to determine available JTAG devices
- [ ] Use FPGA Manager to determine available devices
- [ ] Use Vivado to program bitstream

sommerlukas, 2020-04-28:
What do you think about adding a possibility to program bitstreams programmatically? That might be useful if you use the TaPaSCo API as a base for some other runtime, e.g. OpenMP and OpenCL.

Ideally, the new `tapasco-load-bitstream` would use the same API to reduce code duplication.

jahofmann, 2020-04-28:
Sure, it remains a Python script with a `if __main__` block. You can simply include that script in your program and use the functionality as desired.

sommerlukas, 2020-04-28:
I was thinking that it might make sense to include that into the TaPaSCo API, so it can be used from the same programming languages as the regular API for transferring data and launching jobs;

jahofmann, 2020-04-28:
You mean the toolflow or the runtime?

We could also include it into the runtime, but that's certainly less convenient considering the languages used. However, certainly possible and worth considering.

sommerlukas, 2020-04-28:
Yes, I meant the runtime. 

I'm aware that running all these steps, in particular the external commands, from C/C++/Rust is more challenging than from a simple bash/Python script, but it might be interesting to integrate it into the runtime next to all the other functions for the host-part of the application.

jahofmann, 2020-04-28:
Sure, I'm up for it... the runtime is about 90% ready, only waiting for the release of the current milestone.

closed 2024-04-15 by cahz