esa-tu-darmstadt/tapasco #190: Update copyright headers

jahofmann, 2020-05-06:
Right now there are many random copyright headers all over TaPaSCo. I never really bothered to update them, so most of them still reference e.g. Jens, or even TPC. We should chose one common format and add this format to all files to keep it sane.

Does anyone have a proposal for such a header.

LGPL would suggest:
`TaPaSCo System on Chip generator and runtime
Copyright (C) 2020 Embedded Systems and Applications, TU Darmstadt

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Lesser General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Lesser General Public License for more details.

You should have received a copy of the GNU Lesser General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
`

lukasmweber, 2020-05-07:
The LGPL Header seems fine to me. When we remove the Names from the file, it makes sense to set up a CONTRIBUTERS.md in the root folder. In this file we can also include student's and external contributors (eg. Micha Ober  for his F1 Contribution). In turn we can then remove the names in the files themselves.

jahofmann, 2020-05-11:
Lukas and I should've updated most of the headers. The missing few will be updated as we go.

We should not make sure that we check for these headers in pull request reviews.

closed 2020-05-11 by jahofmann