esa-tu-darmstadt/tapasco #192: Prepare for Release 2020.4

jahofmann, 2020-05-07:
- [x] Prepare Release Branch
- [ ] Fix the copyright headers from #190 
- [ ] Update version specifiers & Check Readme Downloadlinks
- [ ] Test Pynq Bitstream Generation
- [ ] Test Zedboard Bitstream Generation
- [ ] Test VC709 Bitstream Generation
- [ ] Test VC709 Default Tools (memcheck, benchmark, debug, arrayinit, arraysum, arrayupdate)
- [ ] Add VC709 benchmark file
- [ ] Test ZC706 Bitstream Generation
- [ ] Test ZC706 Default Tools (memcheck, benchmark, debug, arrayinit, arraysum, arrayupdate)
- [ ] Add ZC706 benchmark file
- [ ] Test NetFPGA SUME Bitstream Generation
- [ ] Test NetFPGA SUME Default Tools (memcheck, benchmark, debug, arrayinit, arraysum, 
arrayupdate)
- [ ] Add NetFPGA SUME benchmark file
- [ ] Test AU250 Bitstream Generation
- [ ] Test Bittware Bitstream Generation
- [ ] Test Ultra96 Bitstream Generation
- [ ] Test Ultra96 Default Tools (memcheck, benchmark, debug, arrayinit, arraysum, 
arrayupdate)
- [ ] Test ZCU102 Bitstream Generation
- [ ] Test ZCU102 Default Tools (memcheck, benchmark, debug, arrayinit, arraysum, 
arrayupdate)
- [ ] Plot benchmarks

jojowi, 2020-05-08:
[vc709.benchmark.log](https://github.com/esa-tu-darmstadt/tapasco/files/4599615/vc709.benchmark.log)
[xupvvh.benchmark.log](https://github.com/esa-tu-darmstadt/tapasco/files/4599616/xupvvh.benchmark.log)

cahz, 2020-05-11:
[zedboard.benchmark.log](https://github.com/esa-tu-darmstadt/tapasco/files/4608571/zedboard.benchmark.log)
[zc706.benchmark.log](https://github.com/esa-tu-darmstadt/tapasco/files/4608572/zc706.benchmark.log)
[zcu102.benchmark.log](https://github.com/esa-tu-darmstadt/tapasco/files/4608574/zcu102.benchmark.log)
[ultra96v2.benchmark.log](https://github.com/esa-tu-darmstadt/tapasco/files/4608575/ultra96v2.benchmark.log)
[pynq.benchmark.log](https://github.com/esa-tu-darmstadt/tapasco/files/4608576/pynq.benchmark.log)

jahofmann, 2020-05-11:
Adds performance Plot of all the Benchmarks and the old 2019.4 ones as well.

The results are not directly comparable as the bitstreams contain a varying amount of counters and run a  different frequencies. We should make a "Benchmark Guide" to make sure that we have the same bitstreams and machines all the time.

[performance.pdf](https://github.com/esa-tu-darmstadt/tapasco/files/4610658/performance.pdf)

jahofmann, 2020-05-11:
Release Notes
--------------------

We are happy to announce TaPaSCo 2020.04 coming right to you with a number of new platforms bringing TaPaSCo to the cloud and adding HBM support.

AWS Support
============
TaPaSCo now supports AWS F1 instances. Start generating bitstreams using our [Guide](https://github.com/esa-tu-darmstadt/tapasco/wiki/TaPaSCo-on-AWS). No changes to your runtime necessary!

HBM Support
============
TaPaSCo now supports HBM enabled FPGAs, namely the XUPVVH from Bittware and the AU280 from Xilinx. Different HBM channels can be attached to the PEs as desired.

MPSoC and Zynq
==============

The driver and bitstream generation for MPSoC and Zynq has been modernized and is now easier to use than ever. This change comes with an entirely new root FS generator.

Bugfixes
=======
In total this release closes 18 [Issues](https://github.com/esa-tu-darmstadt/tapasco/issues?q=is%3Aissue+milestone%3A2020.4+is%3Aclosed) in 317 commits.

closed 2020-05-13 by jahofmann