esa-tu-darmstadt/tapasco #196: Add .editorconfig to project

m-ober, 2020-05-08:
I noticed that some files have mixed indentation, trailing whitespaces etc.

All editors support configuration by `.editorconfig` (some need a plugin, like vim). Thus, I would like to suggest adding an `.editconfig` file to the project, so the most basic formatting settings are specified.

jahofmann, 2020-05-08:
We aim to use clang-format for everything that is supported (e.g. C, C++).

TCL is a bit of an issue, not a lot of tools seem to support autoformatting for it. I'd be happy to find something here.

Regarding Scala @ls has to say how it's handled.

For python we should use something like https://github.com/hhatto/autopep8

Accordingly, I am not sure if an .editorconfig is flexible enough for all the supported styles. For instance, the kernel code uses different formatting compared to the rest of the C/C++ code.

jahofmann, 2020-05-08:
Well, looking at the format at https://editorconfig.org it should at least be flexible enough... Feel free to create one if you like :D

sommerlukas, 2021-04-07:
Closed by #220

PR #220 (2021-04-07): Add .editorconfig to project
Coming up with an `.editorconfig` was harder than expected, because a lot of different styles are used in this project. Hopefully I got most of it covered :wink:  

Closes #196
commit 9dc6a1d7 (2021-04-07): Add .editorconfig to project

Closes #196
closed 2021-04-07 by sommerlukas