esa-tu-darmstadt/tapasco #205: Host and Memory Frequencies for Ultra96v2 wrong

jahofmann, 2020-05-28:
Memory is at 533, should be 333.33 (why did this ever work?!).

Host is only at 100MHz, should be much higher to decrease latencies.

m-ober, 2020-06-04:
Are you sure about the memory frequency? According to all source I can find, 533 seems correct, for instance [1, 2].

[1] https://www.element14.com/community/groups/fpga-group/blog/2018/07/31/lpddr4-timing-parameters-for-zynq-ultrascale-mpsoc-in-vivado
[2] https://www.element14.com/community/community/designcenter/zedboardcommunity/ultra96/ultra96-v2/blog/2019/07/13/lpddr4-memory-differences-between-ultra96-v1-and-ultra96-v2

jahofmann, 2020-06-04:
I just copied this issue over from @cahz in Mattermost and didn't have a look at it myself.

cahz, 2020-06-04:
The timings you refer in [1, 2] are for the memory controller in the hard core. We use the memory frequency for the AXI HP-Ports on the PS system. If you apply a higher frequency than 333.33 MHz you get a critical warning from vivado and timing won't be optimal.

cahz, 2020-06-26:
Fix is merged into the develop branch.

PR #226 (2020-06-26): Ultra96 frequency fix
Changed the the memory interface frequency to the max. allowed 333 Mhz and the host frequency to 250 Mhz. More is possible, but this is a sweet spot for the lowest interrupt latency. 

Updated ultra96v2.benchmark accordingly.

Closes #205
closed 2020-06-26 by cahz