esa-tu-darmstadt/tapasco #212: [tapasco-build-toolflow] paths are hardcoded

Febbe, 2020-06-06:
The paths in tapasco-build-toolflow are hardcoded, this will produce problems with cloning the repository in any folder, even the home directory, since tapasco-build-toolflow expects ~/TaPaSCo as path and git clone will clone the repo into ./tapasco. All of the paths should be relative.

jahofmann, 2020-06-06:
Thank you for the issue, however the script we have in the master doesn not use hardcoded paths:
```bash
#!/bin/sh

if [ -z ${TAPASCO_HOME_TOOLFLOW} ]; then
    echo "Please prepare TaPaSCo first by sourcing tapasco-setup.sh"
else
    cd ${TAPASCO_HOME_TOOLFLOW}/scala
    ./gradlew installDist
fi
```

https://github.com/esa-tu-darmstadt/tapasco/blob/master/toolflow/bin/tapasco-build-toolflow

Febbe, 2020-06-06:
Ok I think I know what happened:
I've done the following:
``` bash
cd git && git clone https://github.com/esa-tu-darmstadt/tapasco.git
cd ../.. && mkdir "TaPaSCo workspace" && cd "TaPaSCo workspace"
cp ../git/tapasco/tapasco-init.sh tapasco-init.sh
tapasco-init.sh
source tapasco-setup.sh
~/git/tapasco/toolflow/bin/tapasco-build-toolflow
```
This resulted into: 
```bash
~/git/tapasco/toolflow/bin/tapasco-build-toolflow: 6: cd: can't cd to ~/TaPaSCo/toolflow/scala
~/git/tapasco/toolflow/bin/tapasco-build-toolflow: 7: ~/git/tapasco/toolflow/bin/tapasco-build-toolflow: ./gradlew: not found
```
Now I think the real cause is, that the generated `tapasco-setup.sh` can't handle spaces in the path.

jahofmann, 2020-06-06:
Looking at your commands, you're not supposed to copy over `tapasco-init.sh` but simply run it from where it is.

Try the following
```bash
../git/tapasco/tapasco-init.sh
source tapasco-setup.sh
~/git/tapasco/toolflow/bin/tapasco-build-toolflow
```

Otherwise the script does not know here your Tapasco is located because you moved it.

Febbe, 2020-06-06:
Then the documentation in https://github.com/esa-tu-darmstadt/tapasco#tapasco-toolflow-setup is misleading. 

> 3. Create or open a folder, which you would like to use as your TaPaSCo workspace. Within this folder, run the TaPaSCo-Initialization-Script tapasco-init.sh which is located in the root-folder of your cloned repo. This will setup your current folder as TAPASCO_WORK_DIR. It will also create the file tapasco-setup.sh within your workdir.

jahofmann, 2020-06-06:
Well, it says run, not copy and run the copy.

Febbe, 2020-06-06:
ok thank you, and sorry for the misunderstanding it works now with space free paths. But with spaces in the path `tapasco-init.sh` failes to create the path variable:
```bash
cd git && git clone https://github.com/esa-tu-darmstadt/tapasco.git
cd ../.. && mkdir "tapasco_space workspace" && cd "tapasco_space workspace"
~/git/tapasco/tapasco-init.sh
echo $TAPASCO_WORK_DIR
```
Produces `~/tapasco_space` but `"~/tapasco_space workspace"` was expected.

sommerlukas, 2020-06-08:
Fixed in #213;

PR #213 (2020-06-08): Allow spaces in the TaPaSCo Home and Workdir
	- Fixes #212
commit 22a80eaa (2020-06-08): Allow spaces in the TaPaSCo Home and Workdir

	- Fixes #212
closed 2020-06-08 by sommerlukas