esa-tu-darmstadt/tapasco #217: Remove platform zcu102revd

cahz, 2020-06-17:
As the platform is not supported in standard vivado versions, this results in unnecessary warnings/errors when doing e.g. ´hls´ without stating a platform. In my opinion we should remove it from TaPaSCo.

jahofmann, 2020-06-17:
Should be okay. We still have one of the ES ZCU102, but as you correctly stated, it's quite annoying to use. Generating the necessary kernel etc is even more of a hassle...

cahz, 2020-06-17:
There isn't much configuration required anyway, so it's easy to recreate it if needed. 

@jojowi is there any use of the SFP-Plugin? Or is it obsolete after your refactoring?

jahofmann, 2020-06-17:
There is no use of it anymore if @jojowi supports it... that's still my ancient hacky version.

commit 336ee4b7 (2020-06-17): Remove zcu102revd

Closes #217
closed 2020-06-17 by cahz