esa-tu-darmstadt/tapasco #219: Tapasco fails successfully

cahz, 2020-06-17:
Tapasco does not detect a failure of vivado if bitstream and reports are generated. The difference to #174 is that reports are written before the error.

To reproduce this behavior, add `error 1` in `toolflow/vivado/platform/common/platform.tcl`, after `tapasco::call_plugins "post-bitstream"`

```
[18:58:46 <Thread-6: VivadoComposer> INFO] ERROR: [Vivado 12-1345] Error(s) found during DRC. Bitgen not run.
[18:58:46 <Thread-6: VivadoComposer> INFO] 0 Infos, 328 Warnings, 25 Critical Warnings and 3 Errors encountered.
[18:58:46 <Thread-6: VivadoComposer> INFO] write_bitstream failed
[18:58:46 <Thread-6: VivadoComposer> INFO] write_bitstream: Time (s): cpu = 00:00:32 ; elapsed = 00:00:19 . Memory (MB): peak = 8139.797 ; gain = 426.531 ; free physical = 14690 ; free virtual = 34574
[18:58:46 <Thread-6: VivadoComposer> INFO] ERROR: [Common 17-39] 'write_bitstream' failed due to earlier errors.
[18:58:46 <Thread-6: VivadoComposer> INFO] 
[18:58:46 <Thread-6: VivadoComposer> INFO]     while executing
[18:58:46 <Thread-6: VivadoComposer> INFO] "write_bitstream -force "${bitstreamname}.bit""
[18:58:46 <Thread-6: VivadoComposer> INFO]     (procedure "platform::generate" line 96)
[18:58:46 <Thread-6: VivadoComposer> INFO]     invoked from within
[18:58:46 <Thread-6: VivadoComposer> INFO] "platform::generate"
[18:58:46 <Thread-6: VivadoComposer> INFO]     (file "[redacted].tcl" line 160)
[18:58:46 <pool-1-thread-1: VivadoComposer> ERROR] Writing Bitstream failed after 0:00:17 (Total Elapsed: 0:53:14)
[18:58:46 <pool-1-thread-1: VivadoComposer> INFO] Vivado finished successfully for 0x77aebb1e[counter x 16] with axi4mm@[redacted][F=150.000], WNS: 0.091, resulting file is here: '[redacted]--counter_16--150.0.bit'
[18:58:46 <pool-1-thread-1: ComposeTask> INFO] compose run [counter x 16]@150.000 MHz for axi4mm@[redacted] finished, result: Success, produced file: '[redacted]--counter_16--150.0.bit', logfile: '[redacted]--counter_16--150.0.log', utilization report: '[redacted]/counter/016/150.0/utilization.txt', timing report: '[redacted]/counter/016/150.0/timing.txt'
[18:58:46 <main: Compose$> INFO] all Compose tasks finished
[18:58:46 <main: Tapasco$> INFO] TaPaSCo finished successfully

sommerlukas, 2020-06-18:
IIRC, TaPaSCo considers a run to be successful if all reports and a bitstream are generated. In this case, TaPaSCo does not search the log for errors messages, to avoid too many false negatives in cases where the actual process was successful, but Vivado fails afterwards for any non-obvious reason. 

We could reconstruct that process to check the logs for errors, even if reports & bitstream are present, but that might lead to more false negatives. Alternatively, we could have the TCL delete the bitstream, in case a **relevant** post-bitstream step failed. @cahz @jahofmann What do you think?

jahofmann, 2020-06-19:
At the same time we could maybe add another state between Pass and Fail for those cases when routing fails with a WNS but we still consider it most likely good. Maybe make the output Yellow and add a warning that the bitstream might fail.

cahz, 2020-06-19:
The problem in my case was a failed DRC run. Due to some problem, the PCIe interface was not constrained. DRC errors typically result in broken bitstreams.
I think WNS should not cause failures in this step.

m-ober, 2020-06-19:
> TaPaSCo does not search the log for errors messages, to avoid too many false negatives

Does searching for `ERROR:` really yield that much false negatives? From my experience, when an error message appears, no usable output is produced.

I like the idea of @jahofmann to display a warning, like "<Warning> The log contains error messages, bitstream may be unusable. Please check the logfile: .../xyz.log"

cahz, 2020-06-20:
Do we have any case where Vivado returns an error code, but the bitstream is correct anyway? I can't remember the reason why we cannot just forward the error code from Vivado. I think it could be enough to check for the report files only if Vivado finishes successfully.

lukasmweber, 2020-07-08:
To reiterate this issue, so that @mhrtmnn can start on solving it:
- When -0.3 <= WNS <= 0 -> Warning Message, that the bitstream might not work (yellow)
- When Tapasco finishes successfully (Return Code 0), check whether all required Files (bitstream, reports) are actually generated, if not check error logs and forward the message.

Is there anything else that we need to be aware of?

cahz, 2020-07-16:
Maybe include #209 and most importantly, always fail if Vivado fails (return code != 0).

mhrtmnn, 2020-07-16:
@cahz in your PR #214 you added the WPWS check to `platform.tcl`. There seems to be another timing violation check on scala side [here](https://github.com/esa-tu-darmstadt/tapasco/blob/fb289341464694eb63b59d432b2d527490d27307/toolflow/scala/src/main/scala/tapasco/activity/composers/VivadoComposer.scala#L164) (currently only checks WNS) which is invoked after Vivado finishes.

Should we extend this scala function to additionally parse and verify the WPWS value from the timing report, or would the check in TCL be sufficient?

cahz, 2020-07-16:
For good style we should also implement the check in the scala part.

mhrtmnn, 2020-07-16:
ok I extended the function.

How about we delete the bitstream prior to running vivado, then we add a check [here](https://github.com/esa-tu-darmstadt/tapasco/blob/fb289341464694eb63b59d432b2d527490d27307/toolflow/scala/src/main/scala/tapasco/activity/composers/VivadoComposer.scala#L121) if the bitstream file exists or not. This should fix the error described in this issue.
Is this acceptable, or should the bitstream be preserved (eg in case of a dry run)?

closed 2020-10-07 by jojowi