esa-tu-darmstadt/tapasco #222: Add HBM feature

cahz, 2020-06-19:
As more supported plattforms have support for HBM, we should have a common feature for using HBM.

The following two use cases come in my mind:
1. Use HBM as main memory (DMA etc.)
2. Use HBM as additional local memory for PEs

Case 1 is especially important for platforms without DRAM (U50).

jojowi, 2020-06-19:
I think this is also related to #181

jahofmann, 2020-06-19:
This is all part of the changes in #181. Afterwards it should be very easy to support a variety of memory configurations with different kinds of accessors.

cahz, 2020-06-19:
Okay. I thought #181 is more about the runtime / software part. If the hardware / tcl stuff is part of that issue, we can close this.