esa-tu-darmstadt/tapasco #235: [FR] please avoid using namespace in headers

7FM, 2020-08-12:
I have encountered compiler erros dependent on the position of the tapasco.hpp include when compiling a testbench using the opencv headers: <opencv2/highgui/highgui.hpp> <opencv2/imgproc/imgproc.hpp>
Therefore I started looking at the tapasco.hpp and noticed bad practice of having a using namespace inside this header(for more info see: https://stackoverflow.com/questions/5849457/using-namespace-in-c-headers#5849668)
Even if this was not the cause of the compiler errors(it is a #define NONE in tapasco_types.h) I think this might be something worth fixing, especially because this is the main header file which will always be used with tapasco. But as I just had a look at those headers I dont know what side effects it would cause on the exisiting code base.

jahofmann, 2020-08-12:
You're absolutely right, and this issue will be fixed with the NewRuntime. Feel free to have a look at the new version at https://github.com/esa-tu-darmstadt/tapasco/blob/feature/NewRuntime/runtime/libtapasco/src/tapasco.hpp

That branch is also already somewhat stable, so feel free to use it.

7FM, 2020-08-12:
good to know I will give it a try, thanks!

closed 2020-08-12 by 7FM