esa-tu-darmstadt/tapasco #263: Rewrite counter example without SystemC

cahz, 2021-02-01:
Vitis HLS has [dropped support](https://forums.xilinx.com/t5/High-Level-Synthesis-HLS/C-Synthesis-on-Vitis-HLS-with-systemC/m-p/1136129/highlight/true#M21495) for SystemC, therefore a rewrite is needed.

BabarZKhan, 2021-04-19:
Details in the README file:

https://github.com/BabarZKhan/tapasco_counter_versions

cahz, 2021-04-19:
Thanks for your effort!

Apparently, there is no good solution possible with HLS. My tests have shown that `ap_wait` does introduce a delay of around 2n+2. However this is not defined and could change in future versions of Vitis. Therefore I am going to replace the counter by a version written in an HDL.

BabarZKhan, 2021-04-19:
Yes I agree.

I would also not consider `ap_wait()` as an equivalent to  SystemC `wait()` .  Also, `ap_wait()` cannot be used for multiple clock cycles like `wait()`. Therefore I don't know why the Xilinx documentation places  `ap_wait()`  as equivalent to  `wait()`. This is factually wrong. 

On the other hand  `ap_wait_n(X)` is theoretically a good equivalent to SystemC `wait()` but as you have seen on the simulation, it doubles the clock cycles as well. 

Having said that, I will still follow on that for future Vitis version.

cahz, 2021-04-19:
See cac2b6e2ba37da998a7e3b3c6bf021052078c743

closed 2021-04-19 by cahz