esa-tu-darmstadt/tapasco #264: Reintroduce tapasco-debug

cahz, 2021-02-01:
´tapasco-debug´ has proven quite useful. However, it is not included in the new runtime.

jahofmann, 2021-02-01:
Yes, that issue is part of the original PR. I decided to not port the old tapasco-debug as that one was quite flaky and failed easily, e.g. by changing the terminal size.

The new version could use a proper TUI library like https://github.com/fdehau/tui-rs

I have a few changes to the runtime aligned together with fixes for NetFPGA SUME which simplify writing a tool like tapasco-debug by allowing `unsafe` access to PE memory.

jahofmann, 2021-02-01:
Maybe we could write a list of desired features:
 * Change the menu: Show all PEs + Platform components. Number or cursor selects desired one.
 * Pressing enter on the item allows access, e.g.:
   - Peek + Poke memory of that component
   - Show register values for PEs
   - Show status registers for INTc and DMA

I guess that is the main functionality of Tapasco Debug?! Anything else that is needed? I would drop support for the latency test as that is already handled by the benchmark.

cahz, 2021-02-01:
For my typical usage, I need the following features:
* PE/memory list and address map (could be combined into a single screen)
* PE register values
* Dump local memory

As interrupt controller and DMA status information is platform dependend, maybe we can move that into a separate application?

jahofmann, 2021-02-01:
If the interrupt controller and DMA are properly registered in the status core, the debug application can simply show the correct information depending on the platform.

sajjadtamimi, 2021-02-01:
On the existing tapasco-debug, users are able to see only the first 5 registers. 

One option that might be useful is to be able to see more register values (more than 5 of them) if users need it

jahofmann, 2021-02-01:
As TaPaSCo does not know the number of parameter we could use a default of e.g. the first 3 and add +/- to change the number of shown parameters.

BabarZKhan, 2021-02-01:
> For my typical usage, I need the following features:
> 
> * PE/memory list and address map (could be combined into a single screen)
> * PE register values
> * Dump local memory
> 
> As interrupt controller and DMA status information is platform dependend, maybe we can move that into a separate application?

ditto for me ! This is also my typical usage. I use TaPaSCo new version but in parallel with the old version. The old version I use only for debug purposes.  So I apparently miss this feature in new version.

cahz, 2022-05-17:
@zyno42 Could you please update or remove the documentation (under `documentation/tapasco-debug.md`)?

closed 2024-04-22 by cahz