esa-tu-darmstadt/tapasco #268: [Question] Why is libtapasco.a over 67MB large?

forflo, 2021-03-02:
Steps to reproduce:

- System: Ubuntu 18.04.
- Cargo installed via apt not rustup.
- All other tapasco dependencies properly installed after `apt-get update && apt-get upgrade`.
- Install commands:
  - `git clone <tapasco>`
  - `cd && mkdir -p tapasco-workspace && cd tapasco-workspace && ../tapasco/tapasco-init.sh`
  - `cd && cd tapasco-workspace && . tapasco-setup.sh && tapasco-build-libs --skip_driver`
  - `cd && cd tapasco-workspace && cd build* && cpack -G DEB && sudo dpkg -i *.deb`
- `du -h /usr/lib/libtapasco.a` => `67M`

Is LTO not run automatically when building release?
Looking into the script `tapasco-build-libs` shows me that apparently the default is `--mode` release.

How can i reduce the size of the static library. In previous versions of tapasco I was able to build a static libtapasco with only 140K.

jahofmann, 2021-03-02:
First of all: Please don't use ancient Rust versions provided by apt, if at all possible.

For the newest Rust version this should be about 37MB by default. If that is still too much for a static library, you can follow many steps shown e.g. in https://github.com/johnthagen/min-sized-rust.

For instance, stripping would reduce that size down to 26MB. Disabling LTO would reduce the size further to about 17M, costing some speed, though, as LTO adds some info to the library for linking with the executable later down the line.

One question, though: Why do you care about the size of a static library? Especially if it is not exactly big anyway. What is the resulting binary size after linking with LTO? For instance, one of my bigger projects using libtapasco is only 2.7M big after LTO.

forflo, 2021-03-02:
> First of all: Please don't use ancient Rust versions provided by apt, if at all possible.

I can try. Why does it matter? (Besides size reduction.)

jahofmann, 2021-03-02:
It's the recommended way of installation as the compiler improves quite rapidly in many aspects. Although 18.04 comes with 1.47 in the updates repository, which is at least somewhat recent.