esa-tu-darmstadt/tapasco #270: Crash after cancelling TaPaSCo-Job

jojowi, 2021-03-08:
Steps to reproduce:
 - Start program with long running TaPaSCo-Job
 - Cancel program while job is running
 - Run program again -> Crash (in this case the VM)
For me this happens most of the time (but not always). The crash sometimes occurs directly at program start, sometimes the first few output lines still get processed (number of PEs, ...) and it crashes afterwards. But the crash always occurs **before** the second program run starts a TaPaSCo-Job.

The dmesg output when building TaPaSCo with driver_debug

[image]
[dmesg_output.txt](https://github.com/esa-tu-darmstadt/tapasco/files/6100933/dmesg_output.txt)

cahz, 2021-03-13:
I can further refine this error: it happens when you cancel a running execution and the interrupt comes at a later point in time. The next execution of the host application crashes the system.

jahofmann, 2021-03-31:
This seems to be a problem where the application is not attached to the driver anymore but the eventfd signal is not properly invalidated/checked for invalidation. Specifically the problem should be at https://github.com/esa-tu-darmstadt/tapasco/blob/8b21b7fa7d0c4142402554425b5b9b337d45d20f/runtime/kernel/pcie/pcie_irq.c#L53

A check for 0 of the eventfd should fix the issue.

PR #282 (2021-05-26): Fix crash after cancelling job
Fixes #270
closed 2021-05-26 by wirthjohannes