esa-tu-darmstadt/tapasco #272: [Question] Is it possible to set a per-IP clock frequency?

forflo, 2021-03-29:
Is something like this possible:

```
tapasco compose [ foo x 1 ] @ 30 MHz [ bar x 3 ] @ 10 MHz
```

forflo, 2021-03-29:
From the CLI documentation, this syntax seems to be invalid. However, would it make sense to add a per-IP clock frequency to the CLI? I'm asking because I would like to try to add that in the future (if it is possible).

jahofmann, 2021-03-31:
In theory it is not difficult to add additional frequencies to the design. Right now the clock generation for most (all?) platforms is as follows: Memory Clock -> MMCM -> Design Clock. The PEs are clocked with the Design Clock. Inside the Architecture one could of course use additional MMCMs to generate additional clocks. There is some overhead with additional CDCs, so it will certainly hurt during PnR, but a user might be fine with that.

What needs changing:
 * Extend the parser to parse and forward frequencies
 * Select fastest frequency as design clock
 * Generate slower frequencies in Arch and attach them to the correct PEs
 * Add slower clocks to the inbound and outbound interconnects

All in all not really difficult to get it working. It certainly will get more difficult when you also want to add support to DSEs.