esa-tu-darmstadt/tapasco #286: [Question] How does TaPaSCo handle global variables

forflo, 2021-06-17:
Hi everyone,

my question is best elaborated by way of an example:

```c
int vec1[SIZE];
int vec2[SIZE];

void TLF(int *res) {
  *res = vec1[0] + vec2[0];
}
```

According to [the Vivado HLS manual](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug902-vivado-high-level-synthesis.pdf) (page 311, "Exposing Global Variables as I/O Ports"), it is possible to transfer the host contents of global arrays to an IP using an I/O protocol specified via the `INTERFACE` pragma directive. Hence, in theory, it should be possible to specify AXI4-Master interfaces for `vec1` and `vec2` in the code above and employ TaPaSCo's runtime to do the data shipment.

I am interested in using TaPaSCo in similar ways. Did you experiment with this use case before? If so, is it possible to use globals with TaPaSCo in such a way? If not, what would need to change to support this?

Thank you & Best regards,
Florian

cahz, 2021-06-19:
I have never used global variables with HLS. If it is possible to configure it to use an AXI master port, then it should integrate into the TaPaSCo flow. The generated AXI interface also requires an address register in the AXI lite control interface.

For HLS, TaPaSCo utilizes a template where it sets the configurations. Maybe you have to add some interface configurations here: https://github.com/esa-tu-darmstadt/tapasco/blob/40f8239ec88a5d07ac2c58fd3594baa3f6e07dbc/toolflow/vivado/common/hls.tcl.template#L53

forflo, 2021-06-28:
Update: Although the manual explicitly states that it is possible to map global variables to RTL ports and to give those ports interface semantics with`#pragma HLS INTERFACE`, I cannot get it to work properly. I tested two versions: Vivado HLS (2018.2) and Vitis HLS (2020.2).

closed 2021-06-28 by forflo